react-native-sensors/react-native-sensors #2: Not getting updates at the intervals requested

corymsmith, 2017-01-19:
For some reason I only get very sporadic updates from the gyroscope and accelerometer (using the observable subscription). Any ideas why that might be happening?

DanielMSchmidt, 2017-01-20:
Will get into this tomorrow, can you tell me which platform you are using?

corymsmith, 2017-01-20:
I was using iOS, I submitted a PR to change the docs but it looks like Android takes ms so maybe the best course of action is to update the iOS native code to convert to seconds instead.

DanielMSchmidt, 2017-01-22:
I see the problem, [iOS](https://developer.apple.com/reference/coremotion/cmmotionmanager/1616135-accelerometerupdateinterval) handles this in seconds, [Android](https://developer.android.com/reference/android/hardware/SensorManager.html#registerListener(android.hardware.SensorEventListener, android.hardware.Sensor, int)) in microseconds. I handled the translation for android, but missed out on it for iOS. I will handle this by transforming the value before passing it to the native component.

DanielMSchmidt, 2017-01-22:
@corymsmith By the way, if you share the project you are working on I will add it to a section in the readme.

commit 71c73aea (2017-01-22): Fix time conversation bug. closes #2
closed 2017-01-22 by DanielMSchmidt