react-native-sensors/react-native-sensors #5: '_reactNativeSensors.Accelerometer is not a constructor'

Bobby-hewitt, 2017-02-28:
I keep getting this error when implementing the api and 'reactNativeSensors.decorator is not a function when using the decorator.  Could do with an example to work off, otherwise any help would be appreciated.

DanielMSchmidt, 2017-03-01:
hey, I will have a look at this. Unfortunately I haven't had the time to work on the example a bit more, this is still a bit WIP. I will try to fix it up and then I will test it. Are you using the latest version?

Bobby-hewitt, 2017-03-01:
That's great, thanks. Yep, I'm using the latest version

Bobby-hewitt, 2017-03-01:
Hey, just to let you know I've worked around this by requiring the entire module and then using dot notation to call the constructor:

import  ReactNativeSensors from 'react-native-sensors';
const accelerationObservable = new ReactNativeSensors.Accelerometer({
...

Thanks v. much :)

DanielMSchmidt, 2017-03-01:
Can you show your original code, so I can fix that bug?

malnor, 2017-03-14:
I had the same problem, the original code is the one that was supplied in the example (which by the way contains errors in the manual installation instructions - I'll try to give you a mini pr on that when I have some time over..). The workaround supplied by @Bobby-hewitt works.

Original code:
import { Accelerometer, Gyroscope, Magnetometer } from 'react-native-sensors';

const accelerationObservable = new Accelerometer({
  updateInterval: 100, // defaults to 100ms
});

Working code:
import ReactNativeSensors from 'react-native-sensors';
const accelerationObservable = new ReactNativeSensors.Accelerometer({
  updateInterval: 100, // defaults to 100ms
});

DanielMSchmidt, 2017-03-14:
@malnor Thank you, I will look into this later today! 👍

DanielMSchmidt, 2017-04-09:
This should be fixed by #6

closed 2017-04-09 by DanielMSchmidt