react-native-sensors/react-native-sensors #7: Gyroscope is undefined at first render

Eyyub, 2017-03-03:
Hello,

It seems that `this.props.Gyroscope` is undefined at first `render` maybe because its initialization happens some `ms` after `render`, however I would like to know how should I correctly handle this behavior, something like that maybe ?

```javascript
render() {
   if(this.props.Gyroscope)
     return (<Foo/>);
   else
     return null;
}
```

Thanks,

DanielMSchmidt, 2017-03-03:
Hey, I will have a look into this this weekend and will come up with a (hopefully) good answer

DanielMSchmidt, 2017-04-09:
@Eyyub I am sorry, I haven't had time to look into this issue. What would be your preferred solution:

- Leave it like it is but make it explicit in the documentation
- Render `null` if one of the values is missing
- Add a `renderLoading` prop, which renders a loading screen while one of the values is missing

Eyyub, 2017-04-12:
Hello! IMO I would rather leave it like this and making it explicit in the documentation. Also you could add an example with a `renderLoading` so people would know how they can handle this behavior. But this is only my opinion, it's all about what things a lib should handle and what things should be left to the user's decision :p

DanielMSchmidt, 2017-04-13:
Okay, I see. I will enhance the docs then and also add it to an example. Thanks for your help!

commit c43af997 (2017-12-04): Document sensors being null in README

Fixes #7
closed 2017-12-04 by DanielMSchmidt