react-native-sensors/react-native-sensors #8: throw new RuntimeException("No Gyroscope found");

stephanedemotte, 2017-03-28:
Hi !

Thank for this useful lib

With Android emulator (android 6.0, nexus 5)
I got this error `No Gyroscope found`

Any Idea ?

DanielMSchmidt, 2017-03-28:
Yeah, this means that you don't have the native gyroscope accessible. It should work on a real device though

stephanedemotte, 2017-03-28:
Ok great thanks !

`throw new RuntimeException` make the app crash.

Maybe a alert is more safer ?

DanielMSchmidt, 2017-03-29:
I thought those would get translated into JS errors which one could catch, was I wrong?

DanielMSchmidt, 2017-03-29:
I would need to use RCTMakeError as stated in the docs: https://facebook.github.io/react-native/docs/native-modules-ios.html (and the android equivalent)

tontonrally, 2017-04-03:
Hey,

Agree with the RCTMakeError. But maybe we don't have to throw exception until we try to access gyroscope/accelerometer. I mean, can we just throw exception when we try to call the "startUpdates" by checking if the gyro/accelerometer is available ?

This way, we'll not get exception at the init of the module.

DanielMSchmidt, 2017-04-03:
@tontonrally Can you mail me on danielmschmidt92@gmail.com? I added you to the org, would be cool if you could provide a small demo for UWP if that isn't too hard for you :) or extend the current example

DanielMSchmidt, 2017-04-03:
I see the point, would be open for PRs 👍 I currently try to fix some issues in my drawer lib for react-navigation, these are unfortunately more urgent that fixes in this project, but I have enough time to review and merge stuff ;)

Lleios, 2017-04-06:
I agree to use RCTMakeError but it doesn't exist for Android yet. My solution is just temporary. #10

DanielMSchmidt, 2017-04-09:
I just merged @Lleios PR (Version 2.3.0). This makes the module fail as you try to access the Sensor. Whoever has time is free to make a PR for exchanging this hard error with `RCTMakeError` both on iOS, Android and UWP and we should be fine 👍

DanielMSchmidt, 2017-12-04:
I think it should work with the promise.reject, I mean it shouldn't break the app

closed 2017-12-04 by DanielMSchmidt