react-native-sensors/react-native-sensors #11: [question] Get heading compass

stephanedemotte, 2017-04-05:
Anyone know how to get the heading compass ?

I try a lot of code to convert accelerometer, but nothing work for the moment.

Any idea on that would be very appreciated.

Thanks !

DanielMSchmidt, 2017-04-06:
Can you maybe explain a bit what you mean by heading compass? What are you exactly trying to achieve?

stephanedemotte, 2017-04-06:
Try to get the true nord.
Magnetometer doesn't return any value

Found this lib https://github.com/yonahforst/react-native-heading but maybe react-native-sensor can do the job ?

Chathu94, 2017-10-13:
Try this - https://github.com/vnil/react-native-simple-compass

DanielMSchmidt, 2017-10-13:
That seems pretty nice, maybe we could use it here and move it into the rxjs structure 🤔

closed 2018-05-03 by stale[bot]

pashute, 2018-05-03:
There's a file in the source code called magnetometer. Is that supporting the Magnetic Field Sensor (i.e.) the compass? Is it working or is it just in preparation?

And is this thread which was marked as a feature request and moved by "main" to done 14 hours ago, actually done?  If so, is there documentation for how to use it?

DanielMSchmidt, 2018-05-04:
I am super unsure what Magnetometer actually means, could be that we already support compass usage, I have just never had a use-case for it. @NoaneDan should no more, he added it in #62.

The automation moves the tickets automatically when they are closed to done 🤷‍♂️

NoaneDan, 2018-05-04:
@pashute The magnetometer is functional an is working exactly like the accelerometer and gyroscope.  The sensor reads the ambient geomagnetic field on axes x, y, z. The values are in microtesla and they give you the device's orientation relative to Earth's magnetic field.

DanielMSchmidt, 2018-05-04:
Thank you for taking the time to clairify @NoaneDan 👍

pashute, 2018-05-04:
And how do you calibrate the compass if needed, or is that something
outside the access to the sensor itself?

firofame, 2019-08-08:
This is something more appropriate for your need:

https://github.com/firofame/react-native-compass-heading