react-native-sensors/react-native-sensors #16: background tasks support

tontonrally, 2017-04-10:
Hey there !

I don't know how it works for android & ios, but for the uwp, I already developed it for windows and it seems to work.

Any plan to support background tasks ?

DanielMSchmidt, 2017-04-11:
I am not sure what would be necessary to do in order to send the events when the app is backgrounded, to be honest. I am not even sure if RN supports that generally despite using [headless js](http://facebook.github.io/react-native/releases/0.43/docs/headless-js-android.html#headless-js) on android

dorthwein, 2017-10-30:
react-native-background-geolocation does this but accesses more sensors (thus more battery).  Maybe check out that project by chance?  Implementing this would be outside by skill set unfortunately.

DanielMSchmidt, 2018-03-27:
Thank you for the tip @dorthwein I will take a look 👍

closed 2018-05-03 by stale[bot]