react-native-sensors/react-native-sensors #19: Accelerometer / Gyroscope are reversed

hmenzagh, 2017-05-11:
Everything is in the title !
iOS, Xcode, 0.44 react native, automatic link
`import RNSensors from 'react-native-sensors';
import {WidthPercent} from '../utils/SizeManager';
const { Accelerometer, Gyroscope } = RNSensors;

export default class Game extends Component {

	constructor(props) {
		super(props);
		this.state = ({anim: new Animated.Value(ViewWidth / 2)});
	}

	componentDidMount () {
		this.Gyro = new Accelerometer({
  			updateInterval: 5,
		});
		this.Gyro.subscribe(({x,y,z})=>(this.state.anim.setValue(this.getOffset(x))));
	}`
Thanks !

DanielMSchmidt, 2017-05-12:
Can expand a bit on that? We are currently only wiring up native APIs, do android and ios differ here?

hmenzagh, 2017-05-12:
I didn't have the chance to test on android but when i call a gyro i get accelerometer results (returns to 0 when i don't move) on the other hand when i call a Accelerometer i get the gyro results

lefoy, 2017-07-03:
I can confirm this, the gyroscope and accelerometer are reversed. The gyroscope returns accelerometer results and vice versa.

DanielMSchmidt, 2017-07-03:
Sorry that I didn't respond yet, could you do a PR maybe?

LeBlaaanc, 2017-12-22:
I agree with these findings, gyroscope and accelerometer seems backwards, this happens in expo's library too so I'm not sure if they are using this module.

rickcha, 2018-09-06:
What the.. I've been so long time trying to calculate angles out of Gyroscope module .. I thought there's magical formula behind !!! because all values become zero all the time.

Now I am using Accelerometer to get Gyroscope data!!

EskelCz, 2018-10-22:
@DanielMSchmidt Accelerometer and Gyroscope are still reversed. Can we get this reopened? Thank you.

DanielMSchmidt, 2018-10-22:
@rickcha @EskelCz I am sorry for the inconveniences for you. Please be aware that this is an open source project that 100% relies on me having free time to work on it to gain nothing from it, so it's not as high of a priority for me as I would like it to be.

That being said I would love to review a PR fixing this issue (reviewing is something I can easily do on the side, while developing takes longer time blocks I might not have). Reopening it in the hope to see a PR soon <3

DanielMSchmidt, 2018-10-23:
Okay, I took my vacation time today to double check this, here is my trace:

- Native Code
	- iOS shows [`self->_motionManager startAccelerometerUpdates`](https://github.com/react-native-sensors/react-native-sensors/blob/d0e9431fe82635ee9db95785596f0b5fee81e752/ios/Accelerometer.m#L86) and therefore calls the iOS Framework motion manager directly with the right type
	- Android shows [`(mySensor.getType() == Sensor.TYPE_ACCELEROMETER`](https://github.com/react-native-sensors/react-native-sensors/blob/d0e9431fe82635ee9db95785596f0b5fee81e752/android/src/main/java/com/sensors/Accelerometer.java#L88), so only forwarding the accelerometer values
- Import of Native code also seems right: https://github.com/react-native-sensors/react-native-sensors/blob/d8324ca96302fbd474a44731528fed735ead2520/src/rnsensors.js#L4
- They are also used in the handle with the right assignment: https://github.com/react-native-sensors/react-native-sensors/blob/d8324ca96302fbd474a44731528fed735ead2520/src/rnsensors.js#L15

With this in mind I don't see that React Native Sensors switches the implementation between gyproscope and acceleropmeter. Taking the comment of @spacesuitdiver in account that expo has the same problem I would conclude that this naming seems to be different from common understanding in Android and iOS. I will add a paragraph in  the readme about that to spare future users the time to find out by themselves

EskelCz, 2018-10-23:
@DanielMSchmidt Thanks for the investigation. I think I've figured out source of the confusion. It's simply gravity. People expect the values of accelerometer to be 0, 0, 0 when instead they show 0, -1, 0. That initially looks like rotation data, it sort of behaves that way. I jumped the gun as well, sorry for that.

So what is actually needed is an accelerometer data minus gravity, just the user motion.
iOS seems to have that option: https://developer.apple.com/documentation/coremotion/cmdevicemotion
Android has a gravity sensor that can be subtracted:
https://developer.android.com/guide/topics/sensors/sensors_motion

I'd love to help but I don't know java, nor objective C :(

EskelCz, 2018-10-23:
Or simply expose the gravity vector, the subtraction can be done in javascript

DanielMSchmidt, 2018-10-24:
That seems like a good idea. As this is a breaking change I will bundle it with the next major version, which will be moving over to the next rxjs version

DanielMSchmidt, 2018-11-28:
:tada: This issue has been resolved in version 5.0.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v5.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 88db495b (2018-10-24): docs: explain that Accelerometer and Gyproscope seem to be reversed

Closes #19