react-native-sensors/react-native-sensors #21: Tilt calculation with accelerometer data

Noitidart, 2017-06-04:
Hi there,

From just the accelerometer data (because my devices dont have a gyro) I was trying to get the tilt. I was hoping to run this by you all here.

So my goal is to get the pitch and the roll.

At least for portrait orientation I think this is correct:

        const { x, y, z } = Accelerometer;
        const roll = Math.atan2(y, z) * 57.3
        const pitch = Math.atan2((-1*x) , Math.pow((y * y) + (z * z), 0.5)) * 57.3;

As you folks have experience with sensors I am thinking you must have came across this question some point in time. Does my calculation look correct? I based it off of here https://www.dfrobot.com/wiki/index.php/How_to_Use_a_Three-Axis_Accelerometer_for_Tilt_Sensing

Thank you

macelii, 2017-06-26:
+ 1

DanielMSchmidt, 2017-06-26:
Hey there, no I haven't done stuff like that and I am currently not using this project actively anymore. If you would like to see features added, let's discuss how you may get involved ;)

Noitidart, 2017-06-27:
Thanks @macelii - I used this above method in an app I relased here, its a simple tilt game:

* Android -https://play.google.com/store/apps/details?id=com.flythru
* iOS - https://itunes.apple.com/us/app/flythru/id1247138870?mt=8

Thanks @DanielMSchmidt  - is there any maintained sensors project you could recommend? I wish I did, but I only have time to be a consumer of the modules right now. :(

DanielMSchmidt, 2017-06-27:
@Noitidart There is non with full ios and android support AFAIK, I am not letting it become stale, but I can't drive it forward right now.

Noitidart, 2017-06-27:
Thanks much for the fast reply! That is great news, as long as it doesn't get outdated I will keep using this I really loved the experience it gave me. I dont have enough knowledge of how `useNativeDriver` is implemented, but if I found out I will try to come back and help land this. :)

closed 2017-12-04 by DanielMSchmidt

damialajogun, 2020-01-03:
Hi @Noitidart , I just downloaded your game (it's pretty challenging, but fun!). I'm also trying to figure out a tilt calculation and wanted to share with you in case you have any insight that can help from your experience with your own app :). I'm building a charades-inspired game where the user tilts their phone down if they correctly guess the word on the screen and then a new word shows (just like Heads Up!). What I'm trying to figure out now is the logic to enable the word on the screen to change if the phone is tilted down. The phone will be in landscape orientation so I'm trying to get the logic to work such that the word on the screen changes if the phone is tilted down at least 90 degrees.
Do you think accelerometer can do this? How did you identify what the values should be in your equation? Thank you so much!

Noitidart, 2020-01-03:
Hi @damialajogun! Thank you!

My calculations are here - https://github.com/Noitidart/FlyThru/blob/master/flythru/App/PlayScreen/Hole/index.js#L59

damialajogun, 2020-01-07:
Hi @Noitidart, do you mind explaining what this part of your code does? I've been trying to figure it out but can't seem to understand what it means.
[image]
What is the 'roll' variable being calculated supposed to mean? Does that calculate the tilt angle on the z-axis?

listenexcellence, 2023-03-11:
Hello @damialajogun and @Noitidart, Have you all sorted out this last issue?