react-native-sensors/react-native-sensors #22: Ability to as animation driver with useNativeDriver

Noitidart, 2017-06-04:
Thanks for this awesome module.

I was using it to drive an animation but it is hurting performance. Is there anyway to use it as we use Animation event drivers -  https://medium.com/xebia/linking-animations-to-scroll-position-in-react-native-5c55995f5a6e

This connects to scroll of a scroll view:
```
    const xOffset = new Animated.Value(0);

    const onScroll = Animated.event([{ nativeEvent: { contentOffset: { x: xOffset } } }]);

    <ScrollView scrollEventThrottle={16} onScroll={onScroll}>
        // ...
    </ScrollView>
```

Noitidart, 2018-04-11:
Is this a no go? I was hoping to use this in my game here:

Google Play - https://play.google.com/store/apps/details?id=com.flythru
iOS App Store - https://itunes.apple.com/us/app/flythru/id1247138870
Github - https://github.com/Noitidart/FlyThru/blob/master/flythru/App/PlayScreen/index.js

I was hoping to use like we do for `<ScrollView>`:

https://medium.com/xebia/linking-animations-to-scroll-position-in-react-native-5c55995f5a6e

```
const xOffset = new Animated.Value(0);

const onScroll = Animated.event(
  [{ nativeEvent: { contentOffset: { x: xOffset } } }],
  { useNativeDriver: true }
);

function CardView(props: { children?: ReactElement<*> }) {
  return (
    <ScrollView
      scrollEventThrottle={16}
      onScroll={onScroll}>
    </ScrollView>
  )
}
```

But I wanted to basically tie the movment of an element to the pitch, as we see in this game, as we tilt the "hole" moves left and right.

DanielMSchmidt, 2018-04-11:
Sorry for my lack of response @Noitidart 

I have honestly no idea, but I would love if you do research on this if you would post your findings here. I am more than happy to accept PRs in this direction 👍 

If you need guidance through the internals of this library, feel free to ask :)

Noitidart, 2018-04-11:
Thanks so much Daniel. I actually don't have any experience with native. :( I tried to get involved with native side but it was really confusing for me. I'm a pure web guy that came over to web.

I can show example of how to use this js side though would that help? I can write it up tomorrow (2:56 am here right now haha)

DanielMSchmidt, 2018-04-11:
@Noitidart I understand the JS side you are talking about, but implementing this requires a good understanding of the React Native bridge. I think you will need to read a lot of Objective-C and Java if you want to have this feature. I think this blog post might be a good start: https://facebook.github.io/react-native/blog/2017/02/14/using-native-driver-for-animated.html

Noitidart, 2018-04-11:
Dang, thanks for that link. I struggle with that stuff. :( Maybe I can find some native devs, Ill try to look.

closed 2018-05-18 by stale[bot]

MichaelBai, 2019-06-06:
@Noitidart  I've developed a RN module which extract device motion event to React Native, and it uses native driver to provide high performance. Hope it helps. 
https://github.com/rrd-fe/react-native-motion-event-manager