react-native-sensors/react-native-sensors #24: Magnetometer not supported

j-nolan, 2017-06-07:
Hello,

I'd like to point out that the Magnetometer is not supported. Using it as a decorator will have no effect, and using it explicitly will throw the following error when instanciated:

`undefined is not an object (evaluating 'api.setUpdateInterval')`

I tried to fix it, but it appears that the native implementation is still to be coded.

My suggestion for now is to remove the Magnetometer from the exposed API and the README.md in order to avoid leading users into believing it supported. I'd be happy to make a PR for this.

DanielMSchmidt, 2017-06-07:
You are right, I haven't had the time to finish it. 

If you would like to help, feel free to do a PR for removing this stuff from the exposed API. 

If you would like to add the functionality I would love to give you a walk through on what needs to be done. In general you can easily see that I took the ground work of [react-native-sensor-manager](https://github.com/kprimice/react-native-sensor-manager) and [react-native-motion-manager](https://github.com/pwmckenna/react-native-motion-manager) and extended it with a nicer JS API. You can see the example for [iOS here](https://github.com/react-native-sensors/react-native-sensors/blob/master/ios/Gyroscope.m) and [Android here](https://github.com/react-native-sensors/react-native-sensors/blob/master/android/src/main/java/com/sensors/Gyroscope.java). I don't know if you have experience with windows phone, [the implementation is here](https://github.com/react-native-sensors/react-native-sensors/blob/master/windows/Gyroscope.cs), maybe @tontonrally has some advice on how to do it. Would be okay to leave this part out for the moment, at least from my side.

j-nolan, 2017-06-07:
Hello, 

Thanks for replying. 

I know nothing about C# and don't own any Apple device so I'm afraid I won't be of much help for iOS support. 

Never did native Android either, but I know Java so I can try to support this platform. 

I'll make a PR for the exposed API tonight and I'll see if I can do something for Android (can't promise a deadline though).

lefoy, 2017-07-17:
+1, would also like to be able to use the magnetometer with this plugin

ebouJ, 2018-02-19:
I want to work with the magnetometer so how is this going guys ? If not i'll try to send a PR

DanielMSchmidt, 2018-02-19:
I think we could use your  PR here, would be very glad to review it!

commit 52f3ff7f (2018-03-16): Magnetometer (#62)

Closes #24
closed 2018-03-16 by DanielMSchmidt