react-native-sensors/react-native-sensors #26: iOS and Android results decimal

lefoy, 2017-07-03:
The `X` and `Y` axis doesn't share the same decimals on Android and iOS. This is fairly simple to handle outside of the plugin but I feel like this should be handled by the plugin.

Also the Z axis doesn't return the same values on Android and iOS, but I don't think this is an issue with this plugin.

**Android**
```
Object {x: -0.1454315185546875, y: -0.1233978271484375, z: 9.813604736328125}
Object {x: -0.1454315185546875, y: -0.1233978271484375, z: 9.813604736328125}
Object {x: -0.14489898681640626, y: -0.12292327880859374, z: 9.813818359375}
```
**iOS**
```
Object {x: 0.00493927001953125, y: 0.01236572265625, z: -1.0119964599609375}
Object {x: 0.00518798828125, y: 0.01222076416015625, z: -1.01226806640625}
Object {x: 0.0051116943359375, y: 0.0124603271484375, z: -1.0126296997070312}
```

Thank you!

DanielMSchmidt, 2017-07-03:
Hey, I would be happy to see a PR for that 👍

lefoy, 2017-07-04:
I ended up using another plugin because I needed the magnetometer. I probably won't have time to work on that sorry.

ringcrl, 2018-01-11:
@efoy  What another plugin did you used?

lefoy, 2018-01-11:
@ringcrl I used a slightly modified version of https://github.com/pwmckenna/react-native-motion-manager

closed 2018-04-03 by stale[bot]

deLTreeTech, 2021-01-20:
Thanks @lihue . This really save my time.

I just did to remedy the challenge.

`const az = accelerometer.aZ * (Platform.OS === 'ios' ? 9.8 : 1);`