react-native-sensors/react-native-sensors #28: Check device sensors

darioodiaz, 2017-08-18:
Hi ! there is a way to check if the device has some sensors? My phone dont have gyroscope and I get a error but my app will be use some features with gyro and other without it, react-native-sensors let me check sensors before start?

DanielMSchmidt, 2017-12-04:
I think it should be checkable if you use the rxjs syntax and check if the promise is rejected. I guess, we should also include it somehow in the observable syntax.

Would you rather 

- Always get a `null` from the sensor
- Have a dedicated error prop that you can check
- both

What do you think is the best one to use?

darioodiaz, 2017-12-14:
Thanks for your response @DanielMSchmidt I'm a little busy these days but as sonn as possible I'll check that solution and update this issue, thanks again

dabit1, 2018-02-17:
Hi! I have just created a pull request to solve that. It was a problem to me because I want to know at the beggining if I'm able to use the sensor or not. 

Whilt it is reviewed you can install by `npm install -S github:dabit1/react-native-sensors#master`

The pull request:
[https://github.com/react-native-sensors/react-native-sensors/pull/47](https://github.com/react-native-sensors/react-native-sensors/pull/47)

DanielMSchmidt, 2018-02-19:
We have one problem, that there is no support for this functionality in windows. The person adding windows left the project, so I will remove windows for now.

closed 2018-02-19 by DanielMSchmidt

DanielMSchmidt, 2018-02-19:
@dabit1 @darioodiaz Please check out version `3.0.0-0` and see if it works. Once I got the OK from you I will publish the official `3.0.0` release 👍  Thank you for dedicating time to this project 👍

commit 3cad0927 (2018-02-19): Add the chance to know if the sensors are available (#47)

Closes #28