react-native-sensors/react-native-sensors #29: undefined is not an object (evaluating 'api.setUpdateInterval')

rngadam, 2017-08-25:
Cannot get this working either on my own code (from create-react-native-app version: 1.0.0) or from the example https://github.com/react-native-sensors/RNSensorExample (I had to use react-native upgrade to fix it).

This is the error (both iOS and Android):

[image]

Lesliesibbs, 2017-10-02:
anyone able to get past thus issue? getting the one on ios and android, tried installing packages from scratch and nothing working?

KevinHu2014, 2017-10-06:
same problem here😭

rngadam, 2017-10-06:
workaround for us was to use Expo API instead which supports many of the sensors already:

https://docs.expo.io/versions/latest/sdk/accelerometer.html

DanielMSchmidt, 2017-10-06:
Sorry, I didn't have time to look into this recently. I will try to find a good solution the next days. By the way the simulator has no sensor data, just that you know

ccostel, 2017-10-15:
I got rid of this error by running `react-native link react-native-sensors`. When I first read the getting started I missed that line.

Azmeer-crescent, 2017-11-18:
I am also getting this same error, even after linking ! ran both examples, no luck

Topppy, 2017-11-23:
here is my solution(IOS):

1.  Run `react-native link react-native-sensors` 

2.  quit your server 

3. and restart  using `react-native run-ios`

DanielMSchmidt, 2017-12-04:
I would close this for now, if this error still occurs, please ping me an example project, so that I may test it myself 👍

closed 2017-12-04 by DanielMSchmidt

zygama, 2017-12-08:
I have the same error when i use https://github.com/react-native-sensors/RNSensorExample/blob/master/SensorExample.js :'(

EltonR, 2019-07-04:
Getting the error too...

DanielMSchmidt, 2019-07-04:
@EltonR Could you please open a new issue and provide specific steps you took that got you the error?