react-native-sensors/react-native-sensors #34: Docs outdated?

epeli, 2017-10-11:
It seems that `new Accelerometer()` returns a promise of an observable and not the observable directly which the README indicates.

Greeny7, 2017-10-11:
Yeah, yesterday I ran "npm update" in my project and get some troubles because of it. Now we need something like 
```javascript 
let accelerationObservable; 

new Accelerometer({
    updateInterval: 100, // defaults to 100ms
}).then((result) => {
    accelerationObservable = result;
    accelerationObservable
      .map(({ x, y, z }) => x + y + z)
      .filter(speed => speed > 20)
      .subscribe(speed => console.log(`You moved your phone with ${speed}`));
});

DanielMSchmidt, 2017-10-11:
I think I introduced this in the last release, which was unintentionally breaking. I am very sorry about that. @Greeny7 would it be possible for you to send in a PR for the docs?

epeli, 2017-10-11:
What I was actually wondering was this api change intentional?  I've never seen a js api where `new`  is used to create a custom promise.

DanielMSchmidt, 2017-10-11:
Not really, we just wanted to have the error on the js side so it's catchable.

closed 2017-12-04 by DanielMSchmidt