react-native-sensors/react-native-sensors #41: sensors shut down when lock screen (android)

VincentSuperper, 2017-11-29:


VincentSuperper, 2017-11-30:
i can't get any sensor data when the screen is turned off.

DanielMSchmidt, 2017-12-04:
Thank you for filing this issue, I will try to reproduce it 👍

VincentSuperper, 2017-12-07:
i tried to add a wakelock, still didn't work

VincentSuperper, 2017-12-07:
it works in background but stops when i turn off the screen

DanielMSchmidt, 2017-12-07:
Okay, I will investigate this over the next weeks. Have you tried if there
is the same behavior on iOS?

VincentSuperper, 2017-12-07:
everything is fine in iOS.

VincentSuperper, 2017-12-07:
i know little about Android, i've tried some solutions on the internet, but none of them works perfect.

VincentSuperper, 2017-12-07:
this issue has bothered me for quite a while, really hope to get some help ASAP.

DanielMSchmidt, 2017-12-07:
I did some research and have a folllow up question. So my research said it can only keep up the sensor data if you [don't allow the device to lock](https://stackoverflow.com/questions/25148894/how-to-keep-my-accelerometer-working-on-lock-screen-onpause#comment39150109_25149107), but I don't think one would like to make it the default. So my question is, does the sensor data come back if the device is unlocked again?

VincentSuperper, 2017-12-07:
yes it does come back

DanielMSchmidt, 2017-12-07:
Okay, so what you would like to have is an option to say "I want to disallow the app to get backgrounded", right? I could try to build in a flag `alwaysOn` or `androidAlwaysOn` and use the technique mentioned [here](https://stackoverflow.com/a/25149107/1559386O). To be honest, this is for me quite low on my prio list, but I would feel super honoured if you would take the time to try  a PR yourself. I guess you would need to duplicate the classes and we would need to switch between them in the JS part, based on the flag

VincentSuperper, 2017-12-07:
BTW, I've tried to use Service once, i didn't know if i did it right, and it turns out that it didn't work on every android device in my office

VincentSuperper, 2017-12-07:
when i turned off the screen, it threw an exception like this

I/SensorManagerFlow: registerListenerImpl
                                                               java.lang.Exception
                                                                   at android.hardware.SystemSensorManager.registerListenerImpl(SystemSensorManager.java:118)
                                                                   at android.hardware.SensorManager.registerListener(SensorManager.java:775)
                                                                   at android.hardware.SensorManager.registerListener(SensorManager.java:682)
                                                                   at com.qsyapp.MainActivity$PersistService$1.onReceive(MainActivity.java:86)
                                                                   at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:918)
                                                                   at android.os.Handler.handleCallback(Handler.java:739)
                                                                   at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                   at android.os.Looper.loop(Looper.java:135)
                                                                   at android.app.ActivityThread.main(ActivityThread.java:5569)
                                                                   at java.lang.reflect.Method.invoke(Native Method)
                                                                   at java.lang.reflect.Method.invoke(Method.java:372)
                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:931)
                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:726)

VincentSuperper, 2017-12-08:
no need to disallow the device to lock, just keep the CPU awake

DanielMSchmidt, 2018-03-27:
Is this a duplicate of #16?

closed 2018-05-03 by stale[bot]