react-native-sensors/react-native-sensors #55: Usage with cocoapods creates dependency issues

Michaelvilleneuve, 2018-02-26:
When used with cocoapods this library adds a React depency on top of the initial one and it causes compilation issues. 
Worked fine when switching to normal linking though

DanielMSchmidt, 2018-02-26:
@namqdam could you help here?

maxkomarychev, 2018-03-04:
@Michaelvilleneuve could you please post exact error you have and a sample from your `Podfile`?

KevinDoom, 2018-04-04:
@Michaelvilleneuve this problem used to plague me with many libraries. I recently came across react-native-maps suggested install via pods and it has this little snippet that seems to solve the problem:

```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == "React"
      target.remove_from_project
    end
  end
end
```

For context:

```
target [your target] do

  rn_path = '../node_modules/react-native'
  rn_maps_path = '../node_modules/react-native-maps'
  pod 'yoga', path: "#{rn_path}/ReactCommon/yoga/yoga.podspec"
  pod 'React', path: rn_path, subspecs: [
    'Core',
    'RCTActionSheet',
    'RCTAnimation',
    'RCTGeolocation',
    'RCTImage',
    'RCTLinkingIOS',
    'RCTNetwork',
    'RCTSettings',
    'RCTText',
    'RCTVibration',
    'RCTWebSocket',
  ]

  pod 'react-native-maps', path: rn_maps_path
  pod 'react-native-google-maps', path: rn_maps_path

end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == 'react-native-google-maps'
      target.build_configurations.each do |config|
        config.build_settings['CLANG_ENABLE_MODULES'] = 'No'
      end
    end
    if target.name == "React"
      target.remove_from_project
    end
  end
end
```

I am by no means a Pod expert but it worked for me.

maxkomarychev, 2018-05-04:
@DanielMSchmidt please don't drop. Cocoapods have many advantages over static linking and it's a natural way to add dependencies into ios project. It is also easier to manage multi-environment app.
I used this library with cocoapods with no issues and also

>  this problem used to plague me with many libraries

I never encountered any kind of such issues with any other library.

It's very hard to help without seeing actual error and/or sample project with the problem.

DanielMSchmidt, 2018-05-04:
I don't feel comfortable with supporting this myself, could I just forward issues regarding this to you @maxkomarychev

maxkomarychev, 2018-05-09:
@DanielMSchmidt I would be happy to help. It's hard to help in this particular case since there's no details about the error per se or the actual podfile & project config.

DanielMSchmidt, 2018-05-09:
Great, that is good to know 👍 I will just close this issue for now and I will have you in my head the next time I hear cocoapods. @Michaelvilleneuve If you have more information for us, please open up another issue 👍

closed 2018-05-09 by DanielMSchmidt