react-native-sensors/react-native-sensors #65: Version 3

DanielMSchmidt, 2018-03-16:
Hey there,

I would like to release version 3 with a whole lot of bug fixes and new features. These are the PRs that went into it:

- [Check if sensors are available](https://github.com/react-native-sensors/react-native-sensors/pull/47) by @dabit1
- [Add linking failure message](https://github.com/react-native-sensors/react-native-sensors/pull/61): If you forget to link the native dependencies you now get `"Native modules for sensors not available. Did react-native link run successfully?"` as an error by @simonbengtsson 
- [Fix sensor availability checks](https://github.com/react-native-sensors/react-native-sensors/pull/60) by @maxkomarychev
- [Fix startup API](https://github.com/react-native-sensors/react-native-sensors/pull/56) by @baygeldin 
- [Magnetometer](https://github.com/react-native-sensors/react-native-sensors/pull/62) by @NoaneDan is not merged yet, but it will be included in the 3.0 release

First of all, I would like to thank you a lot for the effort you put into this and for keeping up with my change requests and slow review times. This is my game plan for this release, feel free to give me some feedback on this:

1. Publish version `3.0.0-1` and get feedback from you
2. Wait for Magnetometer to be in master
3. Publish version `3.0.0-2` and get feedback from you
4. Publish version `3.0.0`

I will start releasing the first new version right now. I am a bit dependant on you to give me feedback, please comment here, so that we can get it shipped together 👍

baygeldin, 2018-03-16:
Sounds great! I'll try the npm version soon. Thank you for your work on this.

DanielMSchmidt, 2018-03-16:
I just published `3.0.0-2`, it would be super nice if you could send in feedback until the end of next week. I will release `3.0.0` once I got two 👍 by you :)

dabit1, 2018-03-16:
Yeah looks good! Good job guys

DanielMSchmidt, 2018-03-19:
@baygeldin did you get to it? I would love to hear a second approval before I release it to the world 🙈

baygeldin, 2018-03-19:
@DanielMSchmidt sorry for delay (I've been busy on the weekend). `3.0.0-2` works fine for me (both accelerometer and gyroscope) :)

DanielMSchmidt, 2018-03-20:
Awesome, I will release 3.0.0 later today

DanielMSchmidt, 2018-03-20:
We have a website now: https://react-native-sensors.github.io/

DanielMSchmidt, 2018-04-19:
I actually forgot to publish 3.0.0, I did it yesterday, sorry about the ~1 month delay -.-

closed 2018-04-19 by DanielMSchmidt

tetreault, 2018-08-04:
hey, found this issue after trying to use the library, and running the link command, i still receive `Native modules for sensors not available. Did react-native link run successfully?`. Am new to RN, so am I missing something here? 

Its an empty project essentially, with just a header and then using this sensor library. 

Misc info:    
- "react": "16.4.1",
- "react-native": "0.56.0",
- "react-native-sensors": "^3.0.0"

`react-native link` command success: 
[image]

Error screenshot: 
[image]

`components/RNSensorOutput.js`: 

```
import React, { Component } from "react";
import { Text, View } from "react-native";
import { decorator as sensors } from "react-native-sensors";

class RNSensorOutput {
  // no lifecycle needed
  render() {
    const { sensorsFound, Accelerometer, Gyroscope } = this.props;

    if (!Accelerometer || !Gyroscope) {
      // One of the sensors is still initializing
      return null;
    }

    return (
      <View style={styles.container}>
        <Text style={styles.welcome}>
          {(sensorsFound["Accelerometer"] &&
            `Acceleration has value: ${Accelerometer}`) ||
            "Acceleration is not available"}
          {(sensorsFound["Gyroscope"] && `Gyro has value: ${Gyroscope}`) ||
            "Gyro is not available"}
        </Text>
      </View>
    );
  }
}

export default sensors({
  Accelerometer: {
    updateInterval: 300 // optional
  },
  Gyroscope: true
})(RNSensorOutput);
```

sreubenstone, 2018-12-11:
I've tried this 100x it doesn't work. Seems expo api is better

DanielMSchmidt, 2018-12-11:
@sreubenstone You are free to choose whatever tool you like to :)

If you tell me what you tried and which version of `react-native-sensors` you used I might be able to help 👍

sreubenstone, 2018-12-11:
Sorry I am getting frustrated.  I commend you on creating these tools, though I did not succeed to get it working, I might be making a stupid mistake. But I just kept on getting the native modules error.

DanielMSchmidt, 2018-12-11:
If you create a repository with the error and create an issue with a link to it, I will take a look and try my best to help you 👍

Daniuns, 2019-01-17:
Hey Guys, I am new on React Native, could you helpe me? I have just installed the react-native-sensors(Yarn add --save react-native-sensors), then I imported on my project but I receive this message "Native Modules for sensors not available. Did react-native link run successfully?". I did what Tetreault did above, and I runned the command "react-native link react-native-sensors", but the error 
still persists =( 

I am using the "react-native-sensors": "^5.1.5".

and this is my package.json

{
  "name": "tcc_mobile",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "lodash": "^4.17.11",
    "react": "16.6.1",
    "react-native": "0.57.7",
    "react-native-gesture-handler": "^1.0.10",
    "react-native-sensors": "^5.1.5",
    "react-navigation": "^3.0.8",
    "rxjs": "^6.3.3"
  },
  "devDependencies": {
    "babel-jest": "23.6.0",
    "jest": "23.6.0",
    "metro-react-native-babel-preset": "0.48.5",
    "react-test-renderer": "16.6.1"
  },
  "jest": {
    "preset": "react-native"
  }
}

Thanks ! =)

DanielMSchmidt, 2019-01-17:
@Daniuns Please ask this question on [StackOverflow using the `react-native-sensors` tag](https://stackoverflow.com/questions/tagged/react-native-sensors)