react-native-sensors/react-native-sensors #92: Try bridge for e2e tests

DanielMSchmidt, 2018-03-30:
https://github.com/invertase/bridge

Salakar, 2018-03-30:
Seconded 😀 will give you a hand once our tests have migrated over.

DanielMSchmidt, 2018-03-30:
Awesome, thank you 👍

Salakar, 2018-04-09:
Just a quick update, not forgotten about this :P 

Been a few updates to Bridge and some major performance gains whilst working on RNFirebase tests. I pulled across RN's native timers functionality into NodeJS so everything is now much faster:

[image]

Testing on our lib getting there as well and using it to iron out bugs / streamline Bridge:
[image]

DanielMSchmidt, 2018-04-29:
@Salakar any updates on this? 😇

DanielMSchmidt, 2018-05-19:
@Salakar Did you make progress on your module? Can I help you somehow? I would really love to have some tests 🙈

closed 2018-06-25 by stale[bot]