react-native-sensors/react-native-sensors #97: then is undefined

benstoehr, 2018-04-15:
Hello, 

I will assume there is something small I'm missing but I can't get my accelerometer to work. 

`import React, {Component} from 'react';
import {
    Platform,
    StyleSheet,
    Text,
    View
} from 'react-native';
import {StackNavigator} from 'react-navigation';
import {Accelerometer} from 'react-native-sensors';
import {Button} from 'react-native-elements';

export default class HomeScreen extends React.Component {

    constructor() {

        super();

    }

    componentDidMount() {
        let accelerationObservable;

        new Accelerometer({updateInterval: 200,})
            .then((observable) => {
            console.log("Yep");
        });
    }

    render() {
        return (
            <View>

                <Text>Benjamin!</Text>
            </View>
        )
    }
}`

My error is 'new _reactNativeSensors.Accelerometer({updateInterval:200}).then is undefined'

Thanks

>

DanielMSchmidt, 2018-04-16:
Which version of react-native-sensors are you using?

benstoehr, 2018-04-16:
"^2.4.2"

fchenchen, 2018-04-18:
Hi, I ran into the same issue today. I solved it by using the Sensor API code on the [npm page](https://www.npmjs.com/package/react-native-sensors). 

It seems that `new Accelerometer({updateInterval: 200,})` directly returns an observable instead of a promise in version 2.4.2.

closed 2018-04-29 by DanielMSchmidt