react-native-sensors/react-native-sensors #115: Streamline API by removing promises

DanielMSchmidt, 2018-05-15:
**Is your feature request related to a problem? Please describe.**
Returning a promise from a constructor makes the API harder to use, we can remove this layer of abstraction easily.

**Describe the solution you'd like**
```js
const observable = new Accelerometer(options);
observable.catch(err => {
})
```

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

maxkomarychev, 2018-05-15:
should rather have static method to check which are available

DanielMSchmidt, 2018-05-15:
Why would you prefer this?

DanielMSchmidt, 2018-08-11:
Still gonna do it when there is time :)