react-native-sensors/react-native-sensors #122: Subscribe to Sensor while app is closed

TimMolenaar, 2018-05-22:
To begin with: the accelerometer from the react-native-sensors repo works like a charm when the app is running in the foreground or when the app is still open in the background. However, based on the documentation I am not able to figure out whether a certain feature is available, namely persistently subscribe to the accelerometer-sensor when the app is closed. Is it possible to subscribe to a sensor such that it continues to measure when an app is closed?

DanielMSchmidt, 2018-05-22:
I have never tried that, would love to see that in action somehow, give it a try please 👍 If it works, let's make sure we document it somewhere for the next generations to come :)

closed 2018-06-28 by stale[bot]

oriharel, 2018-09-29:
Any findings on this issue? I've heard of an Objective-C object called "CMPedometer" which can be used to query all historic motion events. So I don't believe that the app behaves as usual in the background (I have a painful experience with background tasks in iOS) but when launching your app, you can query this object, get all the data that was stored while your app was closed and do stuff with it. question is - is this library provides access to CMPedometer?

oriharel, 2018-09-29:
Also - are you talking about Android or iOS? my comment was about iOS.