react-native-sensors/react-native-sensors #126: Error when build iOS

duongtungls, 2018-05-25:
**Describe the bug**
I have error when build on Xcode

```
/node_modules/react-native-sensors/ios/Magnetometer.m:41:27: error: too many arguments to block call, expected 1, have 3
            reject(@"-1", @"Magnetometer is not active", [[NSError alloc] init]);
            ~~~~~~        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/node_modules/react-native-sensors/ios/Magnetometer.m:46:23: error: too many arguments to block call, expected 1, have 3
        reject(@"-1", @"Magnetometer is not available", [[NSError alloc] init]);
        ~~~~~~        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.
```
**To Reproduce**
Steps to reproduce the behavior:
I install library and link, and add code like example

$ npm install react-native-sensors --save
$ react-native link react-native-sensors
pod 'RNSensors', :path => '../node_modules/react-native-sensors'

* React Native Version: [e.g. 55.4]
* Device iPhone 6

Please help me fix/show me how to fix this error.

Thanks

DanielMSchmidt, 2018-05-25:
Thank you for your bug report, I will take a look at it this weekend

/remind me about this bug tomorrow

maxkomarychev, 2018-05-25:
@duongtungls is your react-native installed via pods or static linking?
this may be relevant https://github.com/oblador/react-native-vector-icons/issues/490#issuecomment-308246423

duongtungls, 2018-05-25:
I didn't install react native via pod

maxkomarychev, 2018-05-25:
this may be the problem, if you install `react-native-sensors` with pods and use static linking for `react-native` itself then code of `react-native-sensors` won't see type definitions of `react-native`.

either add `react-native` through cocoapods or use static linking for `react-native-sensors`

eyu0415, 2018-05-27:
and the console says 
"-[NSError init] called; this results in an invalid NSError instance. It will raise an exception in a future release. Please call errorWithDomain:code:userInfo: or initWithDomain:code:userInfo:. This message shown only once."

maxkomarychev, 2018-05-27:
While construct `[[NSError alloc] init]` doesn't really make sense (it may have sense to agree on some convention about error codes to make errors more meaningful) since it doesn't provide any details about the error the originally reported issue have nothing to do with it but a way of integrating `react-native` and other 3rdparty libraries into host project.

@duongtungls please let us know if you managed to resolve your problem, thanks.

DanielMSchmidt, 2018-05-28:
@maxkomarychev It seems like you got this covered, I'll assign you to the ticket 👍 I am currently in the progress of finally getting an iPhone for this project, so by next weekend I might be able to help. If you need me, just ping me ;)

duongtungls, 2018-05-28:
Hi everyone,

Thanks for your help.

Finally I can build for iOS and Android.
Problem reason is `react-native link react-native-sensors` just add more code in Podfile no any change in Xcode project file.
I just need to temporary rename Podfile in ios folder and run `react-native link react-native-sensors` again and rename back Podfile and build success.

maxkomarychev, 2018-05-28:
@duongtungls thanks for your feedback! Glad it helped.

closed 2018-05-28 by maxkomarychev

DanielMSchmidt, 2018-05-28:
@maxkomarychev Sounds like a great idea, could you add an issue for that and document your findings there? I honestly would have no idea to look where I can start :)