react-native-sensors/react-native-sensors #131: iOS: Simulator crash on Initialization

barbsicle, 2018-05-31:
**Describe the bug**
On the iOS simulator, there is no support for sensors. When the API is activated via either method, the entire app crashes with no red box.

After debugging in XCode, I have found the error to be due to this line in `Accelerometer.m` (it is also in the other 2 sensor module files):

```
- (void) isAvailableWithResolver:(RCTPromiseResolveBlock) resolve
                        rejecter:(RCTPromiseRejectBlock) reject {
    if([self->_motionManager isAccelerometerAvailable])
    {
        /* Start the accelerometer if it is not active already */
        if([self->_motionManager isAccelerometerActive] == NO)
        {
            resolve(@YES);
        } else {
            reject(@"-1", @"Acceletometer is not active", [[NSError alloc] init]);
        }
    }
    else
    {
        reject(@"-1", @"Acceletometer is not available", [[NSError alloc] init]);
    }
}
```

The crashes are caused by the construction of error via `[[NSError alloc] init]`. If we change the value to `nil`, the crash is resolved.

@DanielMSchmidt: I can create a PR to resolve this issue, but I am not confident enough of my Obj-C ability - do you think changing the error call to `nil` is a sufficient fix?

**To Reproduce**
1.  Import and copy the sample code (observable or decorator method does not matter)
2.  Run in iOS simulator

**Expected behavior**
App should not crash when sensor is not found.

**Desktop:**

* React Native Version: 0.55.4
* Device: iOS Simulator
* Version: 3.0.0

DanielMSchmidt, 2018-05-31:
Hey @barbsicle Thank you for your issue and for the PR 👍 

While I like your contribution, in general, could you paint me a better picture of what you are trying to accomplish? From what I know it is impossible to run a simulator with sensor data, so what is the point in making the error not explicit? Also if you did not know this yet, where would be a good point in our README / website / docs so that you could have found out that simulators don't provide sensor data earlier?

barbsicle, 2018-05-31:
@DanielMSchmidt: Thanks for the speedy reply. The main objective here is to stop the crash from happening - as it stands, if we were to import the library as-is, we would then not be able to develop at all on the simulator due to the crash.

With regards to the docs, I was not able to find any mention of simulators not working, just one issue I happened across which noted that sensors don't work on Android simulator. I would suggest to put a disclaimer in the README before the API documentation.

Thanks!

DanielMSchmidt, 2018-05-31:
So your goal is to stop the crash so you can develop everything not sensor-related in the simulator? That makes sense, I'll take a look into the PR.

Would you mind doing a small doc PR that mentions the lacking sensor support on all simulators?

maxkomarychev, 2018-05-31:
what was the crash reason?

barbsicle, 2018-05-31:
@maxkomarychev Sorry for the delay, I had to rebuild on XCode.

The error is `Thread 17: EXC_BAD_INSTRUCTION (code=EXC_I386_INVOP, subcode=0x0)`.

Stacktrace is here (Item 8 is what calls `[[NSError alloc] init]`):
[image]

maxkomarychev, 2018-05-31:
@barbsicle ok cool, thanks! I'm sorry I made you rebuild whole thing :)

cindybin333, 2018-08-27:
Hi, this is still happening. Any updates? @maxkomarychev

jan-happy, 2018-09-13:
@DanielMSchmidt here is the fix: [https://github.com/react-native-sensors/react-native-sensors/pull/151](url)