react-native-sensors/react-native-sensors #140: Warnings about modules requiring main queue setup

benjarwar, 2018-06-14:
**Describe the bug**
After recently upgrading to React Native `0.55.4`, I've started seeing similar `YellowBox` warnings for three modules:

> Module Gyroscope requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`.

> Module Accelerometer requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`.

> Module Magnetometer requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`.

**To Reproduce**
Steps to reproduce the behavior:

1.  I'm only using the `Accelerometer` in my project
2. `import { Accelerometer } from "react-native-sensors";` 
3.  See error in Simulator on app start

**Expected behavior**
Not expecting warnings.

**Desktop (please complete the following information):**

* React Native Version: 0.55.4
* Device: iOS Simulator
* Version: 3.0.0

**Additional context**
Here's the same issue [fixed for a React Native component](https://github.com/facebook/react-native/pull/17679).

DanielMSchmidt, 2018-06-14:
Thank you for this detailed bug report, kudos for taking the time!

It seems like you have pretty much sorted out what needs to be done, would you feel okay with doing also the PR? Would love to guide you through the process if you like

maxkomarychev, 2018-06-15:
@benjarwar this is supposed to be fixed by #134 
which version of `react-native-sensors` are you using?

maxkomarychev, 2018-06-15:
@DanielMSchmidt it worth publishing new version with recent fixes

DanielMSchmidt, 2018-07-22:
@maxkomarychev pinging again :)

tetreault, 2018-08-04:
After sorting out my own mistakes with this module I got it working but also receive the yellow box warnings. Not sure how to fix it -- any advice @maxkomarychev @DanielMSchmidt ?

i'm using  `"react-native-sensors": "^3.0.0"` but still have this problem when running my prototype on my iphone X

cyb3rsalih, 2018-08-27:
I only add the react-native-sensors package not even import it. But still see the warnings both for
`Magnetometer` , `Accelerometer` , `Gyroscope`

jan-happy, 2018-09-13:
@DanielMSchmidt: It's supposed to be fixed here: [https://github.com/react-native-sensors/react-native-sensors/pull/134](url)
But that change is not present in 3.0.0

DanielMSchmidt, 2018-09-17:
I now published `3.1.0`

jan-happy, 2018-09-19:
@DanielMSchmidt Thanks a lot 👍