react-native-sensors/react-native-sensors #143: Gyroscope.m causes crash by wrongful initialisation of NSError object.

SintraWorks, 2018-07-19:
**Describe the bug**
In Gyroscope.m (in the method [isAvailableWithResolver:rejecter:]), NSError objects are created through [[NSError alloc] init], rather than using the designated initialiser. This causes its internals not to be initialised properly, and can result in a crash if subsequently methods are called on these objects.

E.g. in RCTUtils.m the function RCTJSErrorFromCodeMessageAndNSError gets passed such an error object and tries to access its localizedDescription, causing a crash.

**To Reproduce**
Cause something to trigger [isAvailableWithResolver:rejecter:] in Gyroscope.m when running in an iOS iPad simulator. The simulator has no gyroscope and will the error path will be followed causing this crash.

**Expected behavior**
There should be no crash.

**Desktop (please complete the following information):**

* React Native Version: [e.g. 0.55.4]
* Device [iOS iPad Simulator]
* Version [?]

• Xcode 9.4.1
• Simulator  e.g: iPad (5th gen), iOS 11.4.

Backtrace:
```
* thread #134, queue = 'com.facebook.react.GyroscopeQueue', stop reason = EXC_BAD_INSTRUCTION (code=EXC_I386_INVOP, subcode=0x0)
    frame #0: 0x00000001157b9f1c CoreFoundation`CFHash + 348
    frame #1: 0x000060c000261880
    frame #2: 0x00000001157bc5c3 CoreFoundation`CFBasicHashFindBucket + 179
    frame #3: 0x00000001157bc4f1 CoreFoundation`CFDictionaryGetValue + 177
    frame #4: 0x00000001158aee6f CoreFoundation`_CFErrorCreateLocalizedDescription + 543
    frame #5: 0x000000010b874c68 Foundation`-[NSError localizedDescription] + 113
  * frame #6: 0x000000011002b0bd React`RCTJSErrorFromCodeMessageAndNSError(code=@"-1", message=@"Gyroscope is not available", error=domain: nil - code: 0) at RCTUtils.m:431
    frame #7: 0x000000010ff42465 React`::__41-[RCTModuleMethod processMethodSignature]_block_invoke_2.218(.block_descriptor=0x000060c001e6a540, code="-1", message="Gyroscope is not available", error=domain: nil - code: 0) at RCTModuleMethod.mm:346
    frame #8: 0x000000010ebcc787 RNSensors`-[Gyroscope isAvailableWithResolver:rejecter:](self=0x000060000023b660, _cmd="isAvailableWithResolver:rejecter:", resolve=0x000000010ff41ae0, reject=0x000000010ff42390) at Gyroscope.m:48
    frame #9: 0x000000010ebcc563 RNSensors`-[Gyroscope resolver:rejecter:](self=0x000060000023b660, _cmd="resolver:rejecter:", resolve=0x000000010ff41ae0, reject=0x000000010ff42390) at Gyroscope.m:30
    frame #10: 0x000000011582eccc CoreFoundation`__invoking___ + 140
    frame #11: 0x000000011582eb84 CoreFoundation`-[NSInvocation invoke] + 308
    frame #12: 0x00000001158478d6 CoreFoundation`-[NSInvocation invokeWithTarget:] + 54
    frame #13: 0x000000010ff44709 React`::-[RCTModuleMethod invokeWithBridge:module:arguments:](self=0x0000608001676cc0, _cmd="invokeWithBridge:module:arguments:", bridge=0x00006000001dad60, module=0x000060000023b660, arguments=2 elements) at RCTModuleMethod.mm:544
    frame #14: 0x000000010ff53abc React`facebook::react::invokeInner(bridge=0x00006000001dad60, moduleData=0x0000608000118150, methodId=0, params=0x00006000002ccba8) at RCTNativeModule.mm:99
    frame #15: 0x000000010ff53571 React`facebook::react::RCTNativeModule::invoke(this=0x00006000002ccb90)::$_0::operator()() const at RCTNativeModule.mm:71
    frame #16: 0x000000010ff534c9 React`::___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke(.block_descriptor=0x00006000002ccb70) at RCTNativeModule.mm:65
    frame #17: 0x00000001172a67ab libdispatch.dylib`_dispatch_call_block_and_release + 12
    frame #18: 0x00000001172a77ec libdispatch.dylib`_dispatch_client_callout + 8
    frame #19: 0x00000001172afbe5 libdispatch.dylib`_dispatch_queue_serial_drain + 1305
    frame #20: 0x00000001172b04fa libdispatch.dylib`_dispatch_queue_invoke + 328
    frame #21: 0x00000001172ac344 libdispatch.dylib`_dispatch_queue_override_invoke + 726
    frame #22: 0x00000001172b336c libdispatch.dylib`_dispatch_root_queue_drain + 664
    frame #23: 0x00000001172b3076 libdispatch.dylib`_dispatch_worker_thread3 + 132
    frame #24: 0x00000001177d2169 libsystem_pthread.dylib`_pthread_wqthread + 1387
    frame #25: 0x00000001177d1be9 libsystem_pthread.dylib`start_wqthread + 13
```

winadiw, 2018-08-02:
Yeah i am trying the Gyroscope on iOS simulator and my app got crashed. So i should test on real device?

DanielMSchmidt, 2018-08-02:
Which version are you using? This should be fixed,  the problem is that you can not start a sensor on a simulator. Therefore we added a way to check if the sensor is available before instanciating it, If I recall correctly via promise.

SintraWorks, 2018-09-09:
I understand. Unfortunately, the code that calls RN sensors is not under our control, and I think it would be better for RNSensors to make a check for the availability of the API before trying to use it, thus avoiding a crash.

> On 2 Aug 2018, at 11:02, Daniel Schmidt <notifications@github.com> wrote:
> 
> Which version are you using? This should be fixed, the problem is that you can not start a sensor on a simulator. Therefore we added a way to check if the sensor is available before instanciating it, If I recall correctly via promise.
> 
>

DanielMSchmidt, 2018-09-09:
I think you are right. I have currently not so much time, so it will be hard for me to come up with a fix. If you would send in a PR I would be more then happy to help 👍

SintraWorks, 2018-09-09:
I’m in a similar predicament. But I’ll see what I can do.

closed 2018-10-16 by stale[bot]