react-native-sensors/react-native-sensors #147: Update android gradle build to use "implementation"

jmichalc, 2018-08-24:
Gradle has deprecated the `compile` keyword in favor of `api` and `implementation`:
https://docs.gradle.org/current/userguide/java_library_plugin.html#sec:java_library_separation

`android/build.gradle` uses `compile` in its `dependencies` for react native. The Getting Started guide in `README.md` also mentions `compile` in step 3 for Android. Gradle reports this with every build:

    WARNING: Configuration 'compile' is obsolete and has been replaced with 'implementation'.

You should just be able to replace `compile` with `implementation` with no other changes.

Thanks

DanielMSchmidt, 2018-08-26:
Hey @jmichalc,

Seems like you already did some research, would you like to do a PR for this? I am more then happy to review ;)

jmichalc, 2018-08-27:
Hi @DanielMSchmidt,

I've created https://github.com/react-native-sensors/react-native-sensors/pull/150 for your review. Thanks!

closed 2018-10-03 by stale[bot]