react-native-sensors/react-native-sensors #162: App crashes in android version 4.4

abakhtiar, 2018-09-25:
**Describe the bug**
After installation of the package, app crashes in android v.4.4.
* React Native Version: "0.55.4"
* Device 'GT-N7105 - 4.4.2'

DanielMSchmidt, 2018-09-25:
Which version are you using?
What does the crash log say?
Can you provide an example app?

abakhtiar, 2018-09-26:
> Which version are you using?
**"react-native-sensors": "^3.1.0"**
> What does the crash log say?
Actually app only crashes without any information neither in terminal nor debug console. 
> Can you provide an example app?
this is the piece of the code that I am using. Please notice that this code works on android > v.5.0 and only crashes on version 4.4.
`import { Accelerometer } from "react-native-sensors";
//...
		new Accelerometer({
			updateInterval: 350 // defaults to 100ms
		})
			.then(observable => {
				accelerationObservable = observable;

				// Normal RxJS functions
				accelerationObservable
					.map(({ x, y, z }) => x + y + z)
					.filter(speed => speed > 0)
					.subscribe(speed => [speedFilter.push(Math.round(speed)), speedFilter.shift(), this.movement()]);
			})
			.catch(error => {
				console.log("The sensor is not available");
			});
`

DanielMSchmidt, 2018-09-26:
I think best I can do is to not that down for others that we need at least version 5 on android. I won't have the time and energy to debug and fix this for older versions I am afraid. I would be glad to review a PR though

closed 2018-11-02 by stale[bot]