react-native-sensors/react-native-sensors #163: Accelerometer interval is ignored on Android

TimMolenaar, 2018-09-27:
**Describe the bug**
For my application, the accelerometer-interval is set to 150ms, which means it should produce (6.6*60*60) = 24000 data points an hour. However, On average the interval is set to approximately 250ms and for other phones than mine this is 400~500ms. 

**To Reproduce**
Steps to reproduce the behavior:

1.  Create a react-native project
2.  Import the react-native-sensors package
3.  Set the interval option to 150
4.  Send the results to a server oid to verify that the interval is not correct

**Expected behavior**
The interval should correspond to the interval provided by the accelerometer-options.

**Desktop (please complete the following information):**

* React Native Version: 0.55.3
* Android (6.0, 7.0, 8.0)
* Version 3.0.0

**Additional context**
It works perfectly fine for iOS. Perhaps it has something to do with the Android OS throttling the sensor interval.

DanielMSchmidt, 2018-10-27:
This seems strange, we just pass the information down to the native components. Reading their documentation the interval is not a hard "you'll get something every x ms" interval, but rather a "please don't give me more then sth every x ms." I think the amount varies with the load on the device (as far as my limited knowledge goes)

TimMolenaar, 2018-10-29:
Their documentation indeed mentions that it is more 'request' than 'demand' (if we're both talking about the delay in https://developer.android.com/guide/topics/sensors/sensors_overview#sensors-monitor). However, it is also mentioned that "as a best practice, you should specify the largest delay that you can because the system typically uses a smaller delay than the one you specify". In my case, it is the opposite: it looks like the system is unable to cope with the requested frequency because it is too small. This is weird, because the default interval is 100ms (faster than the frequency I use). 

Including information about the workings of sensor-subscriptions for Android sounds good to me. Perhaps include a link to the Android documentation in the comments of the example, like this:

```Java
...
new Accelerometer({
  updateInterval: 400 // defaults to 100ms
                      // Might deviate for Android => https://developer.android.com/guide/topics/sensors/sensors_overview#sensors-monitor)
})
...
```

DanielMSchmidt, 2018-11-28:
This should be fixed in the newest `5.x.x`. It might have been a problem with having multiple Accelerometers overwriting each others updateIntervals. The new abstraction deals with that in a global way, so it's more like the real system. 

Please feel free to reopen if you disagree

closed 2018-11-28 by DanielMSchmidt

TimMolenaar, 2018-11-29:
Hi Daniel, thanks for looking into the problem. Could you provide a little more context on this possible bug? When could this potentially occur?

DanielMSchmidt, 2018-11-29:
In pre 5.x.x versions if you have two sensors of the same kind started with different interval values, the last interval value set "wins" and gets used globally

TimMolenaar, 2018-11-29:
We only have one 'constant' that creates accelerometer instances, which can be 'started' and 'stopped' by means of a switch. When the switch is enabled, the application creates a new instance of the accelerometer with a fixed updateInterval. When switched off, the stop() command is passed to the acceleration observable. It may be possible that the 'start' command is executed twice, but that still means that the update interval should be identical, as this is a fixed number for every newly created instance. See the code below. This is the only function that initialises a new accelerometer instance. Is it possible that the updateInterval somehow got overwritten by an external event due to this bug? If I understand you correctly, this is only the case if I initialise another instance of the accelerometer with a different updateInterval, right?

```javascript
export const SensorManager = {
// (...)
async start_accelerometer() {
		log("Start accelerometer initiated");

		// (...)

		new Accelerometer({
			updateInterval: accelerometer.interval // 150ms
		})
			.then(observable => {
				accelerometer.accelerationObservable = observable;

				log("Observable initiated: " + accelerometer.accelerationObservable != null);

				accelerometer.accelerationObservable
					.map(({ x, y, z }) => {
						// Calculate magnitude						
						return Math.sqrt(Math.pow(is_ios ? x * accelerometer.gravitation : x, 2)
							+ Math.pow(is_ios ? y * accelerometer.gravitation : y, 2)
							+ Math.pow(is_ios ? z * accelerometer.gravitation : z, 2)).toFixed(3);
					})
					.subscribe(speed => {
						_save_accelerometer_data(speed, Utilities.getTimeStamp());
					});
			})
			.catch(error => {
				log(error);
			});
}
}
```

DanielMSchmidt, 2018-11-29:
So yeah, if you call it twice and the interval is changed the latest is used. That's a constraint given by the native implementation and to make sure this is correctly abstracted in the future I introduced the changes for version 5. Please consider switching, your code would be way smaller.

TimMolenaar, 2018-11-29:
Allright, but then the bug doesn't seem to apply to this issue, because the updateInterval is never overwritten to a different value.

DanielMSchmidt, 2018-11-29:
> Allright, but then the bug doesn't seem to apply to this issue, because the updateInterval is never overwritten to a different value, as it is a constant of 150. Do you agree with me on that?

yes

> Not sure if you're familiar with the way AsyncStorage is implemented

Not at all, I won't be of any help here :(