react-native-sensors/react-native-sensors #167: Method does not override or implement a method from a supertype

watadarkstar, 2018-11-09:
**Describe the bug**
A clear and concise description of what the bug is.

```
Task :react-native-sensors:compileDebugJavaWithJavac FAILED
/Users/adriancaroll/workspace/github/underwriter-bot-mobile-next/node_modules/react-native-sensors/android/src/main/java/com/sensors/RNSensorsPackage.java:19: error: method does not override or implement a method from a supertype
    @Override
    ^
1 error
```

**To Reproduce**
1. Use latest alpha version of React Native (see below - we are using a fork of it)
2. Install this library
3. Run `react-native run-android`

**Expected behavior**
Compiles.

**Desktop (please complete the following information):**

* React Native Version: git+ssh://git@github.com/listicos/react-native.git#workaround/fling-on-android-P
* Device Android
* Version ^4.1.0

**Additional context**
Related: https://github.com/rebeccahughes/react-native-device-info/issues/213

DanielMSchmidt, 2018-11-10:
Thank you for filing this issue! It seems to me like you already did some research, would you feel comfortable to do a PR fixing this bug?

watadarkstar, 2018-11-10:
@DanielMSchmidt Yeah no problem. I have made a PR here: #168 - please review and merge 😄

closed 2018-11-28 by DanielMSchmidt

DanielMSchmidt, 2018-11-28:
:tada: This issue has been resolved in version 5.0.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v5.0.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #168 (2018-11-28): Fix method does not override or implement a method from a supertype
Closes #167
commit 90b49faf (2018-11-28): Fix method does not override or implement a method from a supertype (#168)

Closes #167