react-native-sensors/react-native-sensors #173: Error when build iOS Sending 'NSNumber * _Nullable' to parameter of incompatible type 'double'

quoc1506, 2018-12-01:
**Describe the bug**
I have error when build on Xcode
`Sending 'NSNumber * _Nullable' to parameter of incompatible type 'double'`

[image]

**To Reproduce**
Steps to reproduce the behavior:
I install library and link, and add code like example

`$ npm i react-native-sensors`
`$ react-native link react-native-sensors`
`pod 'RNSensors', :path => '../node_modules/react-native-sensors'`
`$ pod install `
React Native Version: 0.57.7
Mac OS X: 10.14.1
XCode: 9.4.1
Device iPhone 6s
Please help me fix/show me how to fix this error.

Thanks

DanielMSchmidt, 2018-12-01:
Thank you for filing the bug. I currently have no iOS device on my own so it's rather hard to debug for me. Would you feel confident to give this bug a try?

DanielMSchmidt, 2018-12-01:
Good to know, would you still like to give it a try?

DanielMSchmidt, 2018-12-02:
:tada: This issue has been resolved in version 5.1.4 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v5.1.4)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

DanielMSchmidt, 2018-12-02:
Should be fixed in `5.1.4`

closed 2018-12-02 by DanielMSchmidt

quoc1506, 2018-12-03:
Thanks for super fast support @DanielMSchmidt,
I had fixed this issue by my self, also tried your new commit, but I got new bug, reported here #176
Best regard,

commit 30ba7d87 (2018-12-02): fix(barometer): NSNumber is already a NSNumber

Closes #173
PR #175 (2018-12-02): Barometer: NSNumber is already a NSNumber
Closes #173
commit db43e752 (2018-12-02): fix(barometer): NSNumber is already a NSNumber

Closes #173