react-native-sensors/react-native-sensors #176: Error when run on iOS & android Simulator

quoc1506, 2018-12-03:
Step from #173 

[image]

[image]

iiJoshJenkins, 2018-12-03:
I'm getting the same issue running on a real device.

Was just about to to create an issue when I saw this.

DanielMSchmidt, 2018-12-03:
Can you give me the JS side code you wrote?

iiJoshJenkins, 2018-12-03:
`
import { gyroscope, setUpdateIntervalForType, SensorTypes } from "react-native-sensors";
setUpdateIntervalForType(SensorTypes.gyroscope, 1000);
const gyroscopeSubscription = gyroscope.subscribe(({x,y,z,timestamp}) => {
    console.log(x,y,z,timestamp);
});
`

iiJoshJenkins, 2018-12-03:
```js
import RNSensors from "./src/rnsensors";

export const setUpdateIntervalForType = RNSensors.setUpdateInterval;
```

Fixes the issue, not sure if it's the best way of doing it though.
But because you export `RNSensors` as an object it can't find `setUpdateInterval` apparently.
I may be wrong but after a lot of playing around I got it working at least.

```js
export { setUpdateInterval as setUpdateIntervalForType } from './src/rnsensors';
```

DanielMSchmidt, 2018-12-03:
I think I don't get you, what did actually work for you? Could you send in a PR, so that I can take a look?

quoc1506, 2018-12-03:
Hi @Daniel Schmidt
I’ve cloned your example KêpTheBallGame ang got this issue

DanielMSchmidt, 2018-12-03:
Yeah, this could very well be. There are some workflow issues when working with RN which keeps me from testing a version before pushing it in an efficient manner. 

My proposed fix would be to move [this object](https://github.com/react-native-sensors/react-native-sensors/blob/master/src/rnsensors.js#L25) to just be a bunch of exported functions. This would also mean that [this import](https://github.com/react-native-sensors/react-native-sensors/blob/master/src/sensors.js#L4) would need to change to a `import * as RNSensors` import. Would anyone of you feel comfortable doing this change?

iiJoshJenkins, 2018-12-03:
I can give it a go!

iiJoshJenkins, 2018-12-03:
Should be done, created a PR and tested already.

closed 2018-12-04 by DanielMSchmidt

DanielMSchmidt, 2018-12-12:
:tada: This issue has been resolved in version 5.1.5 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v5.1.5)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: