react-native-sensors/react-native-sensors #191: Can't find module: ./src/decorator

wesdewitte, 2018-12-18:
**Describe the bug**
Upgraded from version 4 to 5.
After upgrading its gives me an error that it can't find module ./src/decorator

**Desktop (please complete the following information):**

* sensors version: 5.1.5
* React Native Version: 0.57.7
* Device: android
* Version 8.1.0

**Additional context**
 Body:
    {"originModulePath":"/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js","targetModuleName":"./src/decorator","message":"Unable to resolve module `./src/decorator` from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`: The module `./src/decorator` could not be found from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`. Indeed, none of these files exist:\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator/index(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`","errors":[{"description":"Unable to resolve module `./src/decorator` from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`: The module `./src/decorator` could not be found from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`. Indeed, none of these files exist:\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator/index(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`"}],"name":"Error","stack":"Error: Unable to resolve module `./src/decorator` from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`: The module `./src/decorator` could not be found from `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/index.js`. Indeed, none of these files exist:\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`\n  * `/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/react-native-sensors/src/decorator/index(.native||.android.js|.native.js|.js|.android.json|.native.json|.json|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx)`\n    at ModuleResolver.resolveDependency (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:120:15)\n    at ResolutionRequest.resolveDependency (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/node-haste/DependencyGraph/ResolutionRequest.js:49:18)\n    at DependencyGraph.resolveDependency (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/node-haste/DependencyGraph.js:219:16)\n    at Object.resolve (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/lib/transformHelpers.js:141:30)\n    at dependencies.map.result (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/DeltaBundler/traverseDependencies.js:373:31)\n    at Array.map (<anonymous>)\n    at resolveDependencies (/Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro/src/DeltaBundler/traverseDependencies.js:369:18)\n    at /Users/wesdewitte/Documents/projects/xg-rn-alarm-prototype/node_modules/metro

DanielMSchmidt, 2018-12-18:
Could this be a caching issue? I can't find a reference to `src/decorator` in this project anymore 🤔

DanielMSchmidt, 2018-12-18:
Awesome 👍

closed 2018-12-18 by DanielMSchmidt