react-native-sensors/react-native-sensors #192: Native modules for sensors not available

sergeyshmakov, 2018-12-19:
Hi guys! Im trying to setup react-native-sensors with latest (0.57.8) version of react-native. I use a guide from official react-native and react-native-sensors websites.

All steps passed and react-native link runned successfully but I have an error with my expo start for android (red screen) with title "Native modules for sensors not available. Did react-native link run successfully?".

With first error at \src\rnsensors.js: 11:4

It looks like react-native-sensors cant find NativeModules. Im a newbie on this and just trying to launch this stuff. Do you have any ideas?

mrigo, 2019-01-23:
I've the same error. Any news?
Thanks!

DanielMSchmidt, 2019-01-23:
Let's put this on StackOverflow. There is already this question, does it fit the setting both of you describe? https://stackoverflow.com/questions/54241096/error-native-modules-for-sensors-not-available-did-react-native-link-run-succe

closed 2019-01-23 by DanielMSchmidt