react-native-sensors/react-native-sensors #196: Is documentation for installation incomplete or outdate?

Josema, 2019-01-08:
```
react-native link react-native-sensors
Looks like you installed react-native globally, maybe you meant react-native-cli?
To fix the issue, run:
npm uninstall -g react-native
npm install -g react-native-cli
```

Actually no idea what react-native link does. The manual version does fit for me. I'm using windows and an Iphone, so I don't have xcode installed.

Any tutorial or guide to use this library with Windows+iOS?

DanielMSchmidt, 2019-01-08:
Thank you for filing the issue. I was under the impression iOS does not work together with windows, because you need OSX to have XCode for iOS development.

Could you give me a link to some resources for setting up windows + iOS for React Native, so that I can look into the issue?

Josema, 2019-01-08:
I'm using Expo. Using Expo client on my iPhone I can just open the project that I'm executing in my computer via expo cli.

DanielMSchmidt, 2019-01-08:
I'm not too familiar with expo, my current understanding is that you can not add native dependencies into it, isn't that true anymore?

Josema, 2019-01-08:
Still true:

> Because you don't build any native code when using Expo to create a project, it's not possible to include custom native modules beyond the React Native APIs and components that are available in the Expo client app.

https://facebook.github.io/react-native/docs/getting-started#caveats

closed 2019-01-08 by Josema