react-native-sensors/react-native-sensors #205: unsubscribe doesnt really unsubscribe in the background

wesdewitte, 2019-01-15:
**Describe the bug**
If i subscribe to for example the accelerometer its gives me all the data i need.
But then i want to unsubscribe because i dont need it anymore.
After i unsubscribe on the accelerometer, i can see in the logs it still prints accelero data.
So the accelerometer is still busy which is a waste of battery.

**To Reproduce**
Steps to reproduce the behavior:

1.  subscribe to accelerometer
2. unsubscribe accelerometer

**Expected behavior**
After the i unsubscribe to the accelerometer i shouldn't be able to see any event coming in from accelerometer.

**Desktop (please complete the following information):**

* React Native Version: 0.57.7
* Device Samsung galaxy 4 Android
* Version 5.1.5

**The logs**
After i unsubscribe i still see these logs coming in:

2019-01-15 15:46:03.567 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563566,"z":9.853337287902832,"y":0.14245499670505524,"x":-0.707486629486084}])
2019-01-15 15:46:03.567 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563567,"z":9.853337287902832,"y":0.14245499670505524,"x":-0.707486629486084}])
2019-01-15 15:46:03.576 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563576,"z":9.854534149169922,"y":0.14365209639072418,"x":-0.7134721279144287}])
2019-01-15 15:46:03.577 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563577,"z":9.854534149169922,"y":0.14365209639072418,"x":-0.7134721279144287}])
2019-01-15 15:46:03.587 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563586,"z":9.854534149169922,"y":0.14245499670505524,"x":-0.7110779285430908}])
2019-01-15 15:46:03.587 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563587,"z":9.854534149169922,"y":0.14245499670505524,"x":-0.7110779285430908}])
2019-01-15 15:46:03.597 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563596,"z":9.860519409179688,"y":0.14006079733371735,"x":-0.7158663272857666}])
2019-01-15 15:46:03.598 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563597,"z":9.860519409179688,"y":0.14006079733371735,"x":-0.7158663272857666}])
2019-01-15 15:46:03.608 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563607,"z":9.861717224121094,"y":0.14365209639072418,"x":-0.7146692276000977}])
2019-01-15 15:46:03.610 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563609,"z":9.861717224121094,"y":0.14365209639072418,"x":-0.7146692276000977}])
2019-01-15 15:46:03.618 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563617,"z":9.860519409179688,"y":0.1388636976480484,"x":-0.7110779285430908}])
2019-01-15 15:46:03.620 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563618,"z":9.860519409179688,"y":0.1388636976480484,"x":-0.7110779285430908}])
2019-01-15 15:46:03.628 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563627,"z":9.855731010437012,"y":0.14006079733371735,"x":-0.7134721279144287}])
2019-01-15 15:46:03.629 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563628,"z":9.855731010437012,"y":0.14006079733371735,"x":-0.7134721279144287}])
2019-01-15 15:46:03.638 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563637,"z":9.855731010437012,"y":0.1412578970193863,"x":-0.707486629486084}])
2019-01-15 15:46:03.640 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563639,"z":9.855731010437012,"y":0.1412578970193863,"x":-0.707486629486084}])
2019-01-15 15:46:03.648 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563647,"z":9.84256362915039,"y":0.1388636976480484,"x":-0.7050924301147461}])
2019-01-15 15:46:03.650 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563649,"z":9.84256362915039,"y":0.1388636976480484,"x":-0.7050924301147461}])
2019-01-15 15:46:03.658 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563657,"z":9.848548889160156,"y":0.13048399984836578,"x":-0.707486629486084}])
2019-01-15 15:46:03.660 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563659,"z":9.848548889160156,"y":0.13048399984836578,"x":-0.707486629486084}])
2019-01-15 15:46:03.668 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563667,"z":9.841365814208984,"y":0.12449848651885986,"x":-0.7098808288574219}])

DanielMSchmidt, 2019-01-21:
I'm sorry for the late reply. This seems like a real bug, do you think you could research into it? My assumption is that we forget to call [stop](https://github.com/react-native-sensors/react-native-sensors/blob/master/src/rnsensors.js#L42) on the JS side, but worst case could be that we don't handle it properly on the native side.

whekwan, 2019-01-22:
@DanielMSchmidt same issues on using the magnetometer. Is it possible to fix it asap?

DanielMSchmidt, 2019-01-22:
Sure, if you send a PR I'll be more than happy to review it as quick as possible.
Unfortunately, I don't have the time to do a PR myself at this point in time.

whekwan, 2019-01-23:
@DanielMSchmidt Since I am not familiar with Objective-C, it is a bit difficult for me to understand the native code very quickly. 
Also, I would like to know how to get the calibrated magnetometer data from the iOS device, the current code is to collect the raw data instead. Thanks!

DanielMSchmidt, 2019-01-23:
@whekwan Well, I am familiar with Objective-C, but I lack the time. I'll try to get it done before my vacation mid February though 💪 

> Also, I would like to know how to get the calibrated magnetometer data from the iOS device, the current code is to collect the raw data instead. Thanks!

Please ask this on Stack Overflow, we have a tag for `react-native-sensors there`

whekwan, 2019-01-24:
> @whekwan Well, I am familiar with Objective-C, but I lack the time. I'll try to get it done before my vacation mid-February though 💪
>
>
> Please ask this on Stack Overflow, we have a tag for [...]

@DanielMSchmidt About the calibrated magnetometer data, I have made a PR for adding the new function to collect it from device-motion data, but the unsubscribe issues are still fixing, please take a look about my PR, thanks!

jdegger, 2019-01-31:
> I'm sorry for the late reply. This seems like a real bug, do you think you could research into it? My assumption is that we forget to call [...]

It seems like there is no missing stop on the JS side and if we follow the path it seems to be correct until it's calling `CMMotionManager stopDeviceMotionUpdates`. We are stuck here debugging the issue ourselves. Help would be very much appreciated.

RWOverdijk, 2019-02-04:
> 2019-01-15 15:46:03.567 23632-23632/com.xguard.my_bodyguard W/unknown:ReactNative: Calling JS function after bridge has been destroyed: RCTDeviceEventEmitter.emit(["Accelerometer",{"timestamp":1547563563566,"z":9.853337287902832,"y":0.14245499670505524,"x":-0.707486629486084}])

The listener is probably not removed properly. Is this also a problem on iOS? And, does this also affect the gyroscope? I might look into this later.

RWOverdijk, 2019-02-04:
@wesdewitte Could you give me your test code? Saves me time once I look into it :)

wesdewitte, 2019-02-13:
Is there already any progress on this issue?

ifokeev, 2019-02-22:
same issue here

fchenchen, 2019-03-11:
Hi, I believe that I found the issue. It is actually on the Javascript side in the sensor.js file. The stop function is never triggerd. Please see the changes below. 

In the original code, it seems like the subscribe function will just return an empty function or the observable will just use the empty function even when it is changed later. I am not sure why. The new way works and stop will be triggered. 

NOTE
I have not fully tested the code, for example, when sensor is not avaible. 
I am not very experienced, so hopefully someone else can test this code and submit a PR. 
Thanks for this fantastic library. Hope this helps. 

```javascript
function createSensorObservable(sensorType) {
  return Observable.create(function subscribe(observer) {
    // this.unsubscribeCallback = () => {};
    this.isSensorAvailable = false;
    this.unsubscribeCallback = () => {
      if (!this.isSensorAvailable) return;
      // console.log('Unsubscribed')
      RNSensors.stop(sensorType);
    };

    RNSensors.isAvailable(sensorType).then(
      () => {
        DeviceEventEmitter.addListener(listenerKeys.get(sensorType), data => {
          observer.next(data);
        });

        // Register the unsubscribe handler
        // this.unsubscribeCallback = () => {
        //   RNSensors.stop(sensorType);
        // };
        this.isSensorAvailable = true;

        // Start the sensor manager
        RNSensors.start(sensorType);
      },
      () => {
        observer.error(`Sensor ${sensorType} is not available`);
      }
    );

    return this.unsubscribeCallback;
  }).pipe(makeSingleton());
}
```

wesdewitte, 2019-03-14:
@fchenchen I've tested your work and it does work indeed nice job.

So if someone can make an PR for this: 

`function createSensorObservable(sensorType) {
  return Observable.create(function subscribe(observer) {

    this.isSensorAvailable = false;

    this.unsubscribeCallback = () => {
      
      if (!this.isSensorAvailable) return;

      // stop the sensor
      RNSensors.stop(sensorType);
    };

    RNSensors.isAvailable(sensorType).then(
      () => {
        DeviceEventEmitter.addListener(listenerKeys.get(sensorType), data => {
          observer.next(data);
        });
        this.isSensorAvailable = true;

          // Start the sensor manager
          RNSensors.start(sensorType);
      },
      () => {
        observer.error(`Sensor ${sensorType} is not available`);
      }
    );

    return this.unsubscribeCallback;
  }).pipe(makeSingleton());
}`

DanielMSchmidt, 2019-03-14:
@wesdewitte you can do a PR by first forking the project. [This article goes over the process in detail](https://help.github.com/en/articles/fork-a-repo)

wesdewitte, 2019-03-21:
Here is the PR https://github.com/react-native-sensors/react-native-sensors/pull/227

closed 2019-04-27 by stale[bot]