react-native-sensors/react-native-sensors #209: Crash: Sensor barometer is not available

mujtabazaidi, 2019-01-23:
If the device does not contain a sensor then the app crashes.
Is there any method to check whether the device has a sensor or not?

Log:
```
01-23 12:28:39.406  1537  1884 W MDMAgentLogger: KioskDB: Could not get the JSON value for KIOSK_CURRENT_CONFIG
01-23 12:28:39.411  1537  1884 E MDMAgentLogger: Exception while getting the kiosk config
01-23 12:28:39.413 13955 13987 E ReactNativeJS: Sensor barometer is not available
```

DanielMSchmidt, 2019-01-23:
Can you explain to me what you mean by "crash"? It should error the observable.

mujtabazaidi, 2019-01-24:
Crash means The app shows red screen. So is there any way that i could check if the barometer exists before subscribing to barometer.

[image]

legion-zver, 2019-01-30:
+ 1

legion-zver, 2019-01-30:
But I'm falling in the release build... Sensor gyroscope is not available!

legion-zver, 2019-01-30:
```

legion-zver, 2019-01-30:
This patch fix my error (use https://github.com/ds300/patch-package)

```
diff --git a/node_modules/react-native-sensors/src/sensors.js b/node_modules/react-native-sensors/src/sensors.js
index 85b9155..d841bb7 100644
--- a/node_modules/react-native-sensors/src/sensors.js
+++ b/node_modules/react-native-sensors/src/sensors.js
@@ -29,7 +29,8 @@ function createSensorObservable(sensorType) {
         RNSensors.start(sensorType);
       },
       () => {
-        observer.error(`Sensor ${sensorType} is not available`);
+        // observer.error(`Sensor ${sensorType} is not available`);
+          console.warn(`Sensor ${sensorType} is not available`);
       }
     );
```

DanielMSchmidt, 2019-03-05:
@mujtabazaidi Can you please show me more code, so that I have more context to work with

jorgengranseth, 2019-03-08:
@DanielMSchmidt I'm getting the same error when trying to initialise the gyroscope in the same way as in the [GryoscopeImage example](https://github.com/react-native-sensors/react-native-sensors/blob/master/examples/GryoscopeImage/App.js). I just try to run the app on the iOS simulator - I suppose it's not supported there.

jorgengranseth, 2019-03-08:
I did som digging in the code and found the [signature for the `.subscribe` function in the RxJS docs](http://reactivex.io/rxjs/class/es6/Observable.js~Observable.html#static-method-create). You can handle the error by passing an error handler:
```
sensors.gyroscope.subscribe(
    ({ x, y, z }) => {
        // subscription logic
    },
    (err) => {
        this.setState({ error = err })
    },
);
```
Does this solution work for you @mujtabazaidi @legion-zver ?

legion-zver, 2019-03-11:
@jorgengranseth error occurs at the time of `import` on devices where the gyroscope is not physically

closed 2019-04-17 by stale[bot]