react-native-sensors/react-native-sensors #217: Gyroscope data

jnystrom, 2019-02-13:
I have my ios device sitting on a table, and printing out the x,y,z coordinates and the values are changing even sitting on a table.  Why would these values change when the device is not moving at all?

DanielMSchmidt, 2019-02-13:
We just send through the information we get from the native APIs, you might want to filter the data for a certain delta.

closed 2019-02-13 by DanielMSchmidt

jnystrom, 2019-02-14:
@DanielMSchmidt ok, but is this something that is normal?  The device sitting on a table but the x,y,z coordinates changing in my subscription handler? I have tried sampling every 500ms and 1 second.

DanielMSchmidt, 2019-02-14:
Yes, totally normal, these measurements are not that accurate on a physical level to stay the same for then entire value

jnystrom, 2019-02-14:
I am trying to write an app to measure where the sun is located, and seems like I am going to need a pretty accurate reading to be pointing in the correct direction.

DanielMSchmidt, 2019-02-14:
Just to be clear, we just forward the data we get from the native side, so you won't get any more accurate data than this.