react-native-sensors/react-native-sensors #219: Pitch, Yaw, roll?

jnystrom, 2019-02-14:
How do I get the pitch, yaw, or roll angles given this lib?  Is there a way to get this data directly, or do I need to go figure out how to calculate this information based on whatever the x,y,z coordinates represent?

aanafalcao, 2019-02-25:
I would like to know the same!
This android package is able to provide orientation: https://github.com/kprimice/react-native-sensor-manager. However, it's updates don't seem to work when the device locks down.

DanielMSchmidt, 2019-03-05:
You need to figure that out based on what the provided sensors give you. The idea of this library is not to become a toolkit for every use case, but to provide an easy way to work with the raw sensor data on the javascript side of things. There seems to be demand for this kind of solution, so if you would write a package that calculates pitch yaw and roll based on react-native-sensors I would gladly link it in the readme 👍

closed 2019-03-05 by DanielMSchmidt

skizzo, 2020-08-03:
`react-native-sensory-manager` doesn't work for RN >= 0.61.5. Is there any way to find out yaw, pitch and roll for Android devices at this point? Thanks!