react-native-sensors/react-native-sensors #220: Heading from magnetometer?

jnystrom, 2019-02-14:
I am trying to get the direction the device is pointing, is there a way to get the heading from this library?

DanielMSchmidt, 2019-02-15:
No, but if you find native apis for this for ios and Android at can add it

jnystrom, 2019-02-15:
now way to get this sort of data, like Expo has:  `Expo.Location.watchHeadingAsync();`

DanielMSchmidt, 2019-02-15:
Maybe you can ask them, I don't know of any

DanielMSchmidt, 2019-03-05:
In general the idea of this library is to expose raw data from the device sensors to the JS side of React Native. This is an additional level of information on top of it, that might be derived from raw sensor data. If you like, feel free to write a package that computes what you need based of the data react-native-sensors provides and we can link it in the readme 👍

closed 2019-03-05 by DanielMSchmidt