react-native-sensors/react-native-sensors #222: barometer sensor not providing any data

benengr, 2019-03-01:
**Describe the bug**
The Observable never fires an event on a Google Pixel 2XL.  It works on an Android Emulator.

**To Reproduce**
```
import React, {Component} from 'react';
import { Text } from 'react-native';
import { barometer } from 'react-native-sensors';

interface Props {}
interface State {
  severity: number,
  subscription: Object | null,
  pressure: number,
}

export default class AddItem extends Component<Props, State> {
  constructor(props: Props) {
    super(props);
    this.state = {
      severity: 2,
      subscription: null,
      pressure: 0.0
    };
  }

  componentDidMount() {
    const subscription = barometer.subscribe(({pressure}) => {
      console.log(`Pressure is ${pressure}`);
      this.setState({pressure});
    });
    this.setState({subscription});
  }

  componentWillUnmount() {

  }
  render() {
    return (<Text>Current Pressure: {this.state.pressure.toFixed(2)}</Text>);
  }
}
```

**Expected behavior**
Pressure reading is updated.

**Device Info**
* React Native Version: 58.5
* Device: Google Pixel 2 XL
* Version: 5.1.7

DanielMSchmidt, 2019-03-05:
This seems odd. Does anyone in @react-native-sensors/maintainers has a Google Pixel and might be able to debug this? Otherwise, I have no idea how to look further into this as I don't own the device and it's, therefore, hard to debug for me

dbanck, 2019-03-05:
I have a Pixel 2 XL. I'll try to reproduce it

dbanck, 2019-03-05:
I'm getting a lot of log output when running a sample app on my Pixel 2 XL:

```
03-05 21:41:01.167 12572 12826 I chatty  : uid=10212(com.sensortest) mqt_js identical 2 lines
03-05 21:41:01.206 12572 12826 I ReactNativeJS: Pressure is 1000.7217407226562
03-05 21:41:01.248 12572 12826 I ReactNativeJS: Pressure is 1000.720947265625
03-05 21:41:01.327 12572 12826 I chatty  : uid=10212(com.sensortest) mqt_js identical 2 lines
03-05 21:41:01.364 12572 12826 I ReactNativeJS: Pressure is 1000.720947265625
03-05 21:41:01.408 12572 12826 I ReactNativeJS: Pressure is 1000.7186279296875
03-05 21:41:01.479 12572 12826 I chatty  : uid=10212(com.sensortest) mqt_js identical 2 lines
03-05 21:41:01.524 12572 12826 I ReactNativeJS: Pressure is 1000.7186279296875
03-05 21:41:01.558 12572 12826 I ReactNativeJS: Pressure is 1000.71533203125
```

And the text component is displaying a pressure changing between `1000.65` and `1000.72`.

Versions I'm using:
* React Native: 0.58.6
* React Native Sensors: 5.1.7

benengr, 2019-03-06:
I will retry this today and check the versions that I'm using.

DanielMSchmidt, 2019-03-06:
Thank you for your help @dbanck, it's highly appreciated

benengr, 2019-03-06:
Sorry for the inconvenience and thank you for your responses.  Turns out that my phone needed to be rebooted and that solved the issue.  I tried other apps before the reboot and they were also not able to read the sensor data.  I'm not sure how my phone got into a state were the barometer wasn't being read, but a reboot solved it.