react-native-sensors/react-native-sensors #235: What is the unit given by the magnetometer?

Lelelo1, 2019-05-01:
What is the unit given by the magnetometer? And is it the same on iOS and android?

Lelelo1, 2019-05-01:
https://forums.developer.apple.com/thread/116222 This library is using should probably not use raw data with [CMMotionManager](https://developer.apple.com/documentation/coremotion/cmmotionmanager?language=objc) on iOS but the calibrated data with [CMDeviceMotion](https://developer.apple.com/documentation/coremotion/cmdevicemotion?language=objc)

Lelelo1, 2019-05-03:
This #210 pull request by whekwan adds readings with CMDeviceMotion and gave me the correct values in μT

Lelelo1, 2019-05-25:
The raw data from magnetometer has device bias -which means that onboard hardware interferes with the magnetometer. It is mentioned [here](https://stackoverflow.com/questions/15380632/in-ios-what-is-the-difference-between-the-magnetic-field-values-from-the-core-l) and I have experienced it as well. Using the raw data to for instance get a compass heading - results in a different heading given depending on which way the device where orientated on start.

cuongtora1996, 2019-06-19:
@Lelelo1  How can u get the compass heading?

Lelelo1, 2019-06-19:
@cuongtora1996 you can use [react-native-simple-compass](https://github.com/vnil/react-native-simple-compass) for that.

cuongtora1996, 2019-06-20:
> @cuongtora1996 you can use [react-native-simple-compass](https://github.com/vnil/react-native-simple-compass) for that.

Tks you

closed 2019-07-27 by stale[bot]