react-native-sensors/react-native-sensors #238: All three sensors reading

kenobi91, 2019-05-29:
**Is your feature request related to a problem? Please describe.**
This isn't a problem, rahter it is a question. Is there a way to get data from all three sensors at once (accelerometer, gyro, magnetometer) or is it just enough to set update interval for all three the same value. Like this

setUpdateIntervalForType(SensorTypes.accelerometer, 100);
setUpdateIntervalForType(SensorTypes.magenetometer, 100);
setUpdateIntervalForType(SensorTypes.gyroscope, 100);

const subscription = accelerometer.subscribe(({ x, y, z, timestamp }) =>
  console.log({ x, y, z, timestamp })
);

const subscription1 = gyroscope.subscribe(({ x, y, z, timestamp }) =>
  console.log({ x, y, z, timestamp })
);

const subscription2 = magenetometer.subscribe(({ x, y, z, timestamp }) =>
  console.log({ x, y, z, timestamp })
);

**Describe the solution you'd like**
For example something like this 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

DanielMSchmidt, 2019-05-29:
Could you put this on StackOverflow with the tag "react-native-sensors" please? https://stackoverflow.com/questions/tagged/react-native-sensors

This I'll answer it there 👍

closed 2019-05-29 by DanielMSchmidt

kenobi91, 2019-06-04:
Hello, could you please answer my question :). I rly think this is something to consider. Thx :)
https://stackoverflow.com/questions/56372582/react-native-sensors-all-three-sensors-reading