react-native-sensors/react-native-sensors #247: Performance Guidance

kirkryan, 2019-06-12:
Hi,

Apologies for the high-level nature of this question, and thanks for creating such a great library. We are using this library to detect acceleration in real-time for cyclists and have found that if we setState each update it seems to saturate the react native bridge entirely - causing touch responsiveness of the app to be severely delayed. What is your guidance for:

1) streaming and listening to the values (should we avoid setState entirely)
2) storing the values over time - are we best to keep a buffer of values in memory and periodically flush to persistent storage i.e. every 2 minutes?

Just keen to get general guidance and best practices from the team in how to handle performance when using this library.

DanielMSchmidt, 2019-06-12:
Hey @kirkryan,

I love that you are enjoying the library :) 
Your use case seems pretty interesting, I would love to see a PR to add it to our [projects list](https://github.com/react-native-sensors/react-native-sensors#cool-projects-using-react-native-sensors).

To your question:

1) You can throttle what the native core sends you by using the `setUpdateIntervalForType(SensorTypes.Accelerometer, 400);` API ([See the docs here](https://react-native-sensors.github.io/docs/Usage.html)) so that the RN bridge is not under fire from the client side. Additionally I would recommend client side throttling with rxjs, [here is a blog post explaining it in detail](https://medium.com/@jvdheijden/rxjs-throttletime-debouncetime-and-audittime-explained-in-examples-c393178458f3)

2) I am unsure I understand the question correctly, what problem are you exactly trying to solve here.

If you would like to go into details here a bit more, please ping me at danielmschmidt92@gmail.com

closed 2019-06-12 by DanielMSchmidt

mangroovie, 2019-06-13:
Same challenges here.
Using redux + redux-persist though for persisting.
- there is a memory buffer on redux for n measurements and after that is filled a server post is triggered and buffer flushed. On failed post persist is reinforced.
500ms interval seems to be the sweet-spot after when UI-starts to lag and even before that some lag is detected

Im investigating now if using rxJS operators to filter non significant values from the sensor streams might solve the problem and keeping a keen eye on this thread now :)