react-native-sensors/react-native-sensors #253: Fix/update to support React Native 0.60+ (error: Native modules for sensors not available)

ceefour, 2019-07-06:
**Describe the bug**
Errors with React Native 0.60+

**To Reproduce**
Steps to reproduce the behavior:

1.  `react-native init` with React Native 0.06+
2.  `npm install --save react-native-sensors`
3.  `react-native link`
4.  (workaroun step that doesn't fully work) `react-native link react-native-sensors`
5.  `react-native run-android`

```
error React Native CLI uses autolinking for native dependencies, but the following modules are linked manually:
  - react-native-sensors (to unlink run: "react-native unlink react-native-sensors")
This is likely happening when upgrading React Native from below 0.60 to 0.60 or above. Going forward, you can unlink this dependency via "react-native unlink <dependency>" and it will be included in your app automatically. If a library isn't compatible with autolinking, disregard this message and notify the library maintainers.
Read more about autolinking: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md  
info JS server already running.
info Installing the app...
:ReactNative:Cannot run program "./node_modules/.bin/react-native" (in directory "C:\Users\ceefour\git\motionrecorder\android\.."): CreateProcess error=2, The system cannot find the file specified
:ReactNative:Automatic import of native modules failed.

> Configure project :app
:ReactNative:Cannot run program "./node_modules/.bin/react-native" (in directory "C:\Users\ceefour\git\motionrecorder\android\.."): CreateProcess error=2, The system cannot find the file specified
:ReactNative:Automatic import of native modules failed.

> Task :react-native-sensors:compileDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings
14 actionable tasks: 2 executed, 12 up-to-date
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Accelerometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Barometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Gyroscope.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Magnetometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Accelerometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Accelerometer
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Barometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Barometer
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Gyroscope.java:69: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Gyroscope
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Magnetometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Magnetometer
8 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-sensors:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s

error Failed to install the app. Make sure you have the Android development environment set up: https://facebook.github.io/react-native/docs/getting-started.html#android-development-environment. Run CLI with --verbose flag for more details.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Accelerometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Barometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Gyroscope.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Magnetometer.java:9: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Accelerometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Accelerometer
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Barometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Barometer
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Gyroscope.java:69: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Gyroscope
C:\Users\ceefour\git\motionrecorder\node_modules\react-native-sensors\android\src\main\java\com\sensors\Magnetometer.java:70: error: cannot find symbol
  private void sendEvent(String eventName, @Nullable WritableMap params) {
                                            ^
  symbol:   class Nullable
  location: class Magnetometer
8 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-sensors:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s

    at checkExecSyncError (child_process.js:629:11)
    at execFileSync (child_process.js:647:13)
    at runOnAllDevices (C:\Users\ceefour\git\motionrecorder\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:75:39)
    at buildAndRun (C:\Users\ceefour\git\motionrecorder\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\index.js:137:41)
    at then.result (C:\Users\ceefour\git\motionrecorder\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\index.js:103:12)
    at process._tickCallback (internal/process/next_tick.js:68:7)
```

[image]

**Expected behavior**

1. Autolinks native module (introduced in React Native 0.60) without `react-native-link react-native-sensors`.
2. Runs well.

**Desktop (please complete the following information):**

* React Native Version: 0.60.0
* Device: Android 9.0 Pie simulator
* Version: 5.3.0

**Additional context**

Related to #252 .

DanielMSchmidt, 2019-07-07:
Thank you for the detailed error report! As you already seem to be in top of the changes made by React Native, would you like to do a PR to fix this?

ceefour, 2019-07-08:
Unfortunately I am just beginner RN developer. I am sticking with RN 0.59 for now. Other modules I am using also not yet upgraded to the new mechanism. It is only a matter of time I guess.

DanielMSchmidt, 2019-07-10:
For android X we could maybe use this tooling, I'll try it somewhen this or next week: https://github.com/mikehardy/jetifier#do_you_need_this

kirkryan, 2019-07-14:
I have this running on Android 0.60.3. You simply need to perform the following:

Edit the build.gradle for react-native-sensors (in node_modules) and add the following dependencies

```
implementation 'androidx.appcompat:appcompat:1.0.2' 
implementation 'androidx.annotation:annotation:1.0.2' 
```

Then, go to the src/main/java.com/sensors and for each of the 4 classes (Accelerometer.java, Barometer.java, Gyroscope.java and Magnetometer.java):

Replace:
`import android.support.annotation.Nullable;`
With:
`import androidx.annotation.Nullable;`

Be warned however that once you do this there's no going back under 0.60.0 without rolling back these changes.

Hope that helps someone!

DanielMSchmidt, 2019-07-15:
Hey,

Could you maybe send in a PR for these changes? If you add a breaking
change comment in the commit body it will automatically make a major bump

marcocastignoli, 2019-07-26:
Do you confirm that react-native-sensors doesn't work on React Native 0.60?

kirkryan, 2019-07-26:
@marcocastignoli - it works now. I submitted a PR that includes the fixes and updated install instructions.

closed 2019-09-01 by stale[bot]

javierjim, 2019-10-28:
Is anyone interested in completing this work and start supporting AndroidX? Note what @kirkryan mentioned about not going back once this change is done. We are using this app in our team, and I could help in making a PR to add support for AndroidX. @DanielMSchmidt, thoughts?