react-native-sensors/react-native-sensors #265: Clean RN project with RNS leads to error

matt3224, 2019-08-20:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:

1.  Spin up a fresh RN project
2.  npm i RNS
3.  Amend the default app.js like the appended file
[App.js.zip](https://github.com/react-native-sensors/react-native-sensors/files/3521635/App.js.zip)
4. run `react-native run-ios`
4.  See error

* React Native Version: 60
* Device iPhone X 12.4 Sim
* Version 5.3.0

DanielMSchmidt, 2019-08-21:
Did you run `react-native link` after `npm install react-native-sensors`?

matt3224, 2019-08-21:
I did. RN 0.60 says to unlink though, but both ways don't work

DanielMSchmidt, 2019-08-28:
Hmm, I'm going to need to look into it. If one of you has time to debug this I'm more than happy to review a PR. :) I'm trying to get some time to look into this this weekend ^^

closed 2019-10-05 by stale[bot]