react-native-sensors/react-native-sensors #266: iOS Memory leak

ControlAltDel01, 2019-08-23:
**Describe the bug**
Memory use is going up when subscribed to a sensor the fast you set the update interval and the more sensors you subscribe to the faster the memory use goes up.
[image]

**To Reproduce**
You can use the repo i created:
1. clone https://bitbucket.org/Something14565/testing-ios-memory-leak/src/master/
2. Start the app on a real iOS device and watch memory use in xcode

or create your own repo:
1. Create a new react native project
2. Install react-native-sensors
3. Set the update interval of accelerometer and gyroscope to 10ms
`setUpdateIntervalForType(SensorTypes.accelerometer, 10);
setUpdateIntervalForType(SensorTypes.gyroscope, 10);`
4. Subscribe to accelerometer and gyroscope
`this.accelerometerSubscription = accelerometer.subscribe((accelerometer) => {
});
this.gyroscopeSubscription = gyroscope.subscribe((gyroscope) => {
});`
5. Start the app on iOS and watch memory use in xcode

**Expected behavior**
The memory use should probably not go up like this.

* React-Native Version: 0.60.5
* Real devices: iPhone 5c and iPhone 6s
* Version: 5.3.0

DanielMSchmidt, 2019-08-25:
Thank you for this thorough bug report! I think I found the root cause: [here](https://github.com/react-native-sensors/react-native-sensors/blob/master/ios/Accelerometer.m#L89) we use `updatesToQueue`. I am not sure if we consume that queue by the handler or not, but I think it's the source of the memory leak. Would you have the time to investigate this issue deeper?

DanielMSchmidt, 2019-08-27:
:tada: This issue has been resolved in version 5.3.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v5.3.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

closed 2019-08-27 by DanielMSchmidt

julioxavierr, 2019-10-01:
Thanks for addressing!

PR #270 (2019-08-27): fix(iOS): Changed iOS to use RCTEventEmitter when sending messages to RN
I am not a iOS developer so feel free to tell me if i did anything wrong. 
This fixes #266 .
sendDeviceEventWithName is deprecated and probably has a memory leak.
What i did was changing it to use RCTEventEmitter sendEventWithName.