react-native-sensors/react-native-sensors #271: Magnetometer accuracy

Sakarisson, 2019-08-27:
Hello, and thanks for creating and maintaining a great library!

I am currently implementing a direction indicator to the navigation app that I am working on. The `magnetometer` sensor is very useful, and it was quite easy to implement this solution. In case the accuracy of the sensor is not good, I want to be able to tell my users that they should calibrate the compass. However, from what I can tell, there is no way to get the sensor accuracy from the library's API. 

The direction indicator is a frequently requested feature in our app, and the accuracy reading would be highly useful in our use-case.

Is this something that is currently on your roadmap?

DanielMSchmidt, 2019-08-27:
Well, the roadmap is community driven. If you are willing to invest some time I am sure we can improve this 👍

zolero, 2019-08-28:
I've been trying to do this as well but like you said you probably need to calibrate it within Javascript with some sort of algorithm to do so. Seems to be quiet hard from what I read online. For me it resulted in leaving out the compass directions and figure out the direction of their current GeoLocation you're moving towards to detect in which direction you go.

Sakarisson, 2019-09-05:
@zolero Thanks for sharing your experience. I am afraid that I will omit the direction indicator for now.

I had a look at implementing this. In Android it's very easy - just include the "accuracy" metric. However, in iOS, from what I can tell, the whole code would need to be refactored. Currently the iOS magnetometer reading gives raw values. These readings do not come with an accuracy metric. The file would need to be refactored to utilize the `CMCalibratedMagneticField` property. I am unfortunately not able to do this, as I have very limited experience in developing for iOS. However, hopefully this information can be helpful.

closed 2019-10-12 by stale[bot]