react-native-sensors/react-native-sensors #278: Crash on start in iOS when RNSensors pod is installed

bentorkington, 2019-09-07:
**Describe the bug**
After installing RNS and running Pod, the iOS project crashes on startup in Accelerometer/RCTRegisterModule():

Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'You must override the `supportedEvents` method of Accelerometer'

**To Reproduce**

1.  react-native init Sensors && cd Sensors
2.  npm install react-native-sensors && react-native link react-native-sensors
3.  cd ios && pod install
4.  Run the project, either from the CLI or XCode

**Expected behavior**
Application starts normally with RNS loaded

**Device: **

* React Native Version: 0.60.5
* Simulator or iOS device
* Version 5.3.2

hoonblizz, 2019-09-09:
Hi, 
`react-native link react-native-sensors` is necessary? 
I believe since RN v0.60, link is replaced to auto link by doing `pod install`.
How about trying again without linking?

bentorkington, 2019-09-09:
Hi,
Same result if I just `npm install react-native-sensors` followed by `pod install`

kylehassett, 2019-09-14:
Does this help at all? Attempting this fix right now.

https://www.gitmemory.com/issue/react-native-sensors/react-native-sensors/272/526697853

kylehassett, 2019-09-14:
I believe the above link has the fix.

Just add the...

`- (NSArray<NSString *> *)supportedEvents {
    return @[@"Magnetometer"];
}`

Just change the string for the sensor to the equivalent for each sensor

bentorkington, 2019-09-14:
Thanks for that. I'd added the supportedEvents which fixed the crash, but still didn't get it to work. Looks like the link you posted has the answer, will try it soon.

bentorkington, 2019-09-14:
This is duplicate of #272, sorry!

closed 2019-09-14 by bentorkington