react-native-sensors/react-native-sensors #281: Got "Thread 1: signal SIGABRT" error on xcode after installing react-native-sensors

rhong7, 2019-09-18:
[image]
It was working successfully when I first initalised the react-native project. But after I installed react-native-sensors, added 'RNSensors' on podfile and ran the command 'pod install' , I'm getting this SIGABRT error. I didn't even import react-native-sensors in my react native code. 

I tried to solve this problem by downgrading react native, searching for SIGABRT error on the internet but still couldn't find solution. spent about two days on this.

"dependencies": {
    "react": "16.8.6",
    "react-native": "0.60.5",
    "react-native-sensors": "^5.3.2"
  },

jayli3n, 2019-09-18:
Me too, hoping for a fix asap.

vohoangankhanh, 2019-09-20:
I had the same problem, switch to 5.3.0 solved this problem

jayli3n, 2019-09-21:
apparently 5.3.0 has unsolved memory leaks etc reading upon previous issues.

W3D3, 2019-10-07:
Tried switching to 5.3.0, issue is still there. Hoping for a fix as well.

michbil, 2019-11-13:
I think this issue should be fixed by [PR](https://github.com/react-native-sensors/react-native-sensors/commit/fcd7338881d9e81658e7f6106bff1155af72c97c). But this change is not present in the latest build on npm (5.3.2)

guopeng1994, 2019-11-14:
which version can use now ?

michbil, 2019-11-14:
I ended using github link instead of npm version in package.json
```  "react-native-sensors": "react-native-sensors/react-native-sensors",```

closed 2019-12-21 by stale[bot]