react-native-sensors/react-native-sensors #293: Questions

cristianoccazinsp, 2019-12-18:
Hello, I'm trying out a few sensor options for RN, I wonder the following:

- Is this maintained?

- Do sensor events fire if the app is in background/suspended? For example, I would like to detect if the user started moving with his phone from a stationary state, and prompt a notification or turn on some services (e.g., GPS) when that happens.

DanielMSchmidt, 2019-12-19:
> Is this maintained?

🤷‍♂ What does maintained mean to you? I think PR reviews are done frequently, but as I currently have no real-world use-case for this library I'm not really adding new sensors or other features all the time

> Do sensor events fire if the app is in background/suspended?

If the React Native bridge works if the app is backgrounded/suspended this should also work. There is no logic to stop the sensor sending data through the bridge when sth is backgrounded.

Hope this helps

cristianoccazinsp, 2019-12-19:
Thanks, that helps.

The fact you answered and that the lastest commits are not 2 years ago I
think counts as maintained :)

cristianoccazinsp, 2019-12-19:
Probably what made me ask it is the fact that there's no RN 0.60 >= references and the install instructions are also for RN < 59.  Nevertheless, gonna give it a try, thanks!

DanielMSchmidt, 2020-01-03:
👍

closed 2020-01-03 by DanielMSchmidt